hzeller/rpi-rgb-led-matrix #1326: 64x64 2/4 and 4/4 of screen work but not 1/4 and 3/4 sections

Stabathane, 2021-07-29:
I have previously gotten the matrix to work with ESP8266 and ESP32 so I know it works. I upgraded to an RPI 3B and got the electrodragon adapter. I bought 2 adapters and solder up the E pin to Pin 4 and Pin 8. I thought I had Pin 8 as seen below but Pin 4 is where I got half of the panel to light up. I have the Hub 75E connector and used the wiring in the image to make my previous versions work. Now for some reason the upper mid and lower fourths of the board light up while the upper and lower middle sections do not. I have tried every combination of command-line parameters with each Pin 4 and Pin 8 enabled. Any ideas?
[image]
[image]

hzeller, 2021-07-29:
Pin 8 is the one you need here (and make sure to only connect to 8 not as well to 4, because that would connect the E to ground).

What are the parameters you give to the libray ? It would need to contain `--led-rows=64`

Stabathane, 2021-07-29:
For the simple square cube demo I ran
`sudo` ./demo -D0 --led-rows=64 --led-cols=64`
For clarity, I have 2 electrodragon adapters, one with Pin 4 enabled and one with Pin 8 enabled.
I tried each one and only Pin 4 gave me that output above.
The board with Pin 8 gave me a black screen.

hzeller, 2021-07-29:
So connecting it with 4 will just short the E output, so this is why you see an output essentially like it was without any E connection.

If the screen is black in the other case, then something is broken with the wiring somewhere. I have not used the electrodragon board myself, so don't know exactly where to look. The Adapter in this project (that I believe the electrodragon is essentially based of) has [this wiring](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing)

Stabathane, 2021-07-29:
I currently have the wiring for Pin 4 enabled like this, the right side silkscreen just shorts P8 to ground, I can try resoldering the board. Will try as fast as I can.
[image]

Stabathane, 2021-07-29:
Update.
It was a touch-up of a bad solder connection for the board with Pin 8 enabled!!!

hzeller, 2021-07-29:
Ha, sometimes having a second look can help :) Have fun with your LED panels!

closed 2021-07-29 by hzeller