hzeller/rpi-rgb-led-matrix #1327: Displaying Static Image on Matrix

conor-f, 2021-08-01:
I have a Pi Zero W, the Adafruit RGB Bonnet and a 64x64 matrix and I'm looking to just display a static image and check if the image should change every few seconds. I'm getting some flickering and I'm kind of confused. I thought as it's an LED display, I'd be able to just display a static, not changing image which wouldn't need any refresh rate thus removing any flicker. Is this an option I'm missing, something not implemented in the library currently, or a physical limitation of the hardware? If so I'd appreciate if someone could link me/give an overview of why because it's surprising to me!

Thanks!

hzeller, 2021-08-01:
Independent of what you show, the display needs constant refresh.

These displays are very simple, so they don't have any framebuffer or even a notion of graylevels, so they have to be externally fed constantly, clocking in the image plane data line-by-line and getting accurate gray-levels with exact microsecond-timed pulses. Towards the end of the README is the [CPU use](https://github.com/hzeller/rpi-rgb-led-matrix#cpu-use) section that talks about it.

A Zero W has the same one-core CPU like the original Pi from 2012, so you will have more likely flickering issues, though a single 64x64 should not be doable.

 It you are using the Adafruit Bonnet instead of the standard wiring used in this project, you don't make use of the hardware timing the library provides; it is highly recommended that you read the [hardware hack needed for Adafruit hats](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) to improve that. In particular for the single-core old CPUs, this can dramatically reduce the flicke, in particular if you see random lines light up brigther every now and then in the current set-up.
It then makes use of a hardware circuit to create the accurate pulses mentioned above.

conor-f, 2021-08-02:
Thanks @hzeller, I have made the hardware modification linked but not the E-Line one as I don't think that's needed on the new bonnet - the process seems to have been made easier with one simple solder blob outlined [here](https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/driving-matrices) in step 3.

I think that the Pi Zero is a little underpowered for what I ultimately want to achieve with this display, so I'll grab a 3B+ or similar to power it!

closed 2021-08-02 by conor-f