hzeller/rpi-rgb-led-matrix #1331: RGBMatrix::Fill vs RGBMatrix::SetPixel

giuseppedevito, 2021-08-11:
Hi *,
I'm using this awesome lib to drive P10 RGB (no SMD) Matrix; pixel mapping is OK and images look good on the display.
Doing some brightness benchmark with demo-main.cc apps (-D 1, -D 4 and so on), I'm noting that if   RGBMatrix::Fill
method is used to to light up pixels (for example a complete white or red display), Brightness is very very high (I premise
that --led-brightness=100 is always used) and the single LED emits its maximum of light;
if RGBMatrix::SetPixel method is used (obviously with an image that fill the display with the same color as the RGBMatrix::Fill
of previous benchmark) the light emitted by every single LED is half the RGBMatrix::Fill one (used --led-brightness is always 100).
Its not a refresh rate concern since tuning --led-pwm-bits or --led-pwm-dither-bits with different values (and subsequent higher
or smaller refresh rates) has been tested without performance changes.
There is a way to improve brightness with RGBMatrix::SetPixel too?

hzeller, 2021-08-11:
mmh, that is odd, SetPixel() and Fill() should be exactly the same w.r.t. the output color and brightness.
In the program where you are using SetPixel() can you also use Fill() and compare with the same program  and settings ?

I could imagine that if you use SetPixel() in a tight loop to set pixels directly on the panel, that intermediate values are displayed and not the final one. I usually would suggest to first fill an off-screen buffer with your content, then SwapOnVSync() it to display.

giuseppedevito, 2021-08-11:
@hzeller  Thank you for the fast reply.
I analyzed the library code (MapColor, DumpToMatrix and so on) and I agree with you that SetPixel and Fill must have the same effects.
To be sure of the emitted LED brightness, I have measured the LUX with a Chroma Meter and I Have almost **1000** LUX with
a complete white image (using setPixel) and almost **13000** LUX with a complete white Fill (Fill(255,255,255)); the high
difference is easily visible by human eyes too. 
In these benchmarks I'm using your demo-main.cc (I'm not using my program) with option:

1) -D 1 -m 0 <path-to>/white.ppm      (where white.ppm has all pixel set to 0xff, verified by the command line prog "od  -x" )
2) -D 4    (The ColorPulseGenerator, modified to not pulse but to call off_screen_canvas_->Fill(255,255,255) and then
                 SwapOnVSync ).

To be complete I write the complete commands:
1) ./demo -D 1 --led-multiplexing=19 --led-show-refresh --led-slowdown-gpio=3 --led-rows=32 --led-cols=128 --led-chain=2 --led-parallel=3 --led-brightness=100 -m 0 /home/gdevito/images/white.ppm

2) ./demo -D 4 --led-multiplexing=19 --led-show-refresh --led-slowdown-gpio=3 --led-rows=32 --led-cols=128 --led-chain=2 --led-parallel=3 --led-brightness=100

As you can see, I have written the PixelMapper 19 to manage LINSN RGB matrixes 16x16 1:1 Scan (Static) with chip 
ICND2038S using RaspPi 3B+ and ElectroDragon board (the red one with 3 flat connectors). 
With some check, I verified that, for this matrix, HUB75 pins R0, G0, B0 manage the first (top-most) 8 pixel rows,
while R1, G1, B1 manage the bottom 8 pixel rows. Address Pins are not used (I verified that, even removing Addresses
cables from the main flat cable, the panel works fine).
Anyway, the mapping is good and I can share the Mapper with the Community.

I verified the output of the FrameBuffer::MapColors method (I thought that CIEMapColor could be the difference) but
colors are translated the same either in the Fill case either in the SetPixel one.

But is the panel brightness set by pulsing the OE pin or by adjusting the PWM color in each ICND2038S output pin?
I'm sorry if my question can seem stupid but I'm not an expert of such electronics.

In a point of the code (framebuffer.cc) I've found double_rows_ = rows/SUB_PANELS where SUB_PANELS is a macro
set to 2. Can you confirm that the value of this macro is yet valid with R0,G0,B0 and R1,G1,B1 of the LED matrixes I'm
using?
Best regards,
Giuseppe

giuseppedevito, 2021-08-12:
Hi *,
to exclude that the problem is in the LED matrix, I have connected a few of 32x32 P6 SMD Panels (with Coreman pixel mapping).
The benchmark results are the same (with RGBMatrix::Fill the brightness is at its maximum value while with 
RGBMatrix::SetPixel the brightness is low).
So, if someone is interested, I think the test can be replicated with whatever LED matrix type.
Have an optimal brightness is important if we want the sun to not wipe out symbols on the screen.

giuseppedevito, 2021-08-12:
I'm noting that the brightness problem is related to the mapping since, if I use another multiplexer (for example Coreman or No
Mapping at all) the brightness is kept high as expected (obviously the image is not intelligible as with my Multiplexer).
I write the mapping using pixel-mover util, with --led-multiplexer = 0 to figure out how visible x and y are mapped on to
physical coordinates on the matrix. In the following xls file (ods is the original) I have written the obtained mapping.
[P16-RGB-1_1StaticScan-Mapping-20210812-rev0.0.1.ods](https://github.com/hzeller/rpi-rgb-led-matrix/files/6974612/P16-RGB-1_1StaticScan-Mapping-20210812-rev0.0.1.ods)
[P16-RGB-1_1StaticScan-Mapping-20210812-rev0.0.1.xls](https://github.com/hzeller/rpi-rgb-led-matrix/files/6974613/P16-RGB-1_1StaticScan-Mapping-20210812-rev0.0.1.xls)

Here my multiplexer:

```
/**
* The matrix is a P16 16x16 RGB (1:1 Scan) that use ICN2038S. 
**/
class GDevP16RGBMultiplexMapper : public MultiplexMapperBase {
public:
  GDevP16RGBMultiplexMapper() : MultiplexMapperBase("GDevP16RGB", 1 /*I set stretch_factor to 1*/) {} 

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
        int xr_offset = ((y / 4) % 2)?0:64;//first or second 8 pix height row
	int xq_offset = (x / 4) * 16;//the x start of the row sub-box
	int xc_offset = (x % 4);//column offset inside the sub-box
	int y_offset = y % 4;
	int xqr_offset = y_offset * 4;//starting x of the sub-box row
	*matrix_y = (y < 8)?0: 16;
	*matrix_x = xr_offset + xq_offset + xc_offset + xqr_offset;
  }
  
  virtual void MapVisibleToMatrix(int matrix_width, int matrix_height, int visible_x, int visible_y,int *matrix_x, int *matrix_y) const {
    const int chained_panel  = visible_x / 16; //the real matrix width is 16
    const int parallel_panel = visible_y / 16; //the real matrix height is 16
    const int within_panel_x = visible_x % 16; //the real matrix width is 16
    const int within_panel_y = visible_y % 16; //the real matrix height is 16
    int new_x, new_y;
    MapSinglePanel(within_panel_x, within_panel_y, &new_x, &new_y);
    *matrix_x = chained_panel  * panel_stretch_factor_*panel_cols_ + new_x;
    *matrix_y = parallel_panel * panel_rows_/panel_stretch_factor_ + new_y;
  }
};
```
Any idea about the problem?

giuseppedevito, 2021-08-12:
In this method:

```
bool RGBMatrix::Impl::ApplyPixelMapper(const PixelMapper *mapper) {
  if (mapper == NULL) return true;
  using internal::PixelDesignatorMap;
  const int old_width = shared_pixel_mapper_->width();
  const int old_height = shared_pixel_mapper_->height();
  int new_width, new_height;
  if (!mapper->GetSizeMapping(old_width, old_height, &new_width, &new_height)) {
    return false;
  }
  PixelDesignatorMap *new_mapper = new PixelDesignatorMap(
    new_width, new_height, shared_pixel_mapper_->GetFillColorBits());
  for (int y = 0; y < new_height; ++y) {
    for (int x = 0; x < new_width; ++x) {
      int orig_x = -1, orig_y = -1;
      mapper->MapVisibleToMatrix(old_width, old_height,
                                 x, y, &orig_x, &orig_y);
      if (orig_x < 0 || orig_y < 0 ||
          orig_x >= old_width || orig_y >= old_height) {
        fprintf(stderr, "Error in PixelMapper: (%d, %d) -> (%d, %d) [range: "
                "%dx%d]\n", x, y, orig_x, orig_y, old_width, old_height);
        continue;
      }
      const internal::PixelDesignator *orig_designator;
      orig_designator = shared_pixel_mapper_->get(orig_x, orig_y);
      *new_mapper->get(x, y) = *orig_designator;
    }
  }
  delete shared_pixel_mapper_;
  shared_pixel_mapper_ = new_mapper;
  return true;
}

```
because I need x E [0, 128] and y E [0,16] to cover the whole 16x16 matrix,
my mapper ofter enters the statement

```
if (orig_x < 0 || orig_y < 0 ||
    orig_x >= old_width || orig_y >= old_height) {
  fprintf(stderr, "Error in PixelMapper: (%d, %d) -> (%d, %d) [range: "
          "%dx%d]\n", x, y, orig_x, orig_y, old_width, old_height);
  continue;
}
```
So the PixelDesignator cannot be updated later at the rows:

```
const internal::PixelDesignator *orig_designator;
orig_designator = shared_pixel_mapper_->get(orig_x, orig_y);
*new_mapper->get(x, y) = *orig_designator;
```

Can be here the problem?

giuseppedevito, 2021-08-12:
I have found a valid solution in #1190  and by downloading and testing @merliongold  #1200 pull request.
Anyway this solution does not work (Segfault) with --led-parallel=3 and --led-chain=4 that I need.
It's frustrating...