hzeller/rpi-rgb-led-matrix #1332: Cannot get a loop working

soulid, 2021-08-17:
Hi,
happy to get an animation working (I am not a Raspberry expert) , but want to show an animation and a picture in a never ending loop. Loop stops after image is shown. How to do so? This is my try:
`sudo examples-api-use/demo -D 0 --led-rows=32 --led-cols=64 --led-chain=2  --led-pixel-mapper=U-mapper --led-daemon
sleep 2
sudo killall demo
sudo ./utils/led-image-viewer  -s -t2   ./img/coordinates.png -C --led-rows=32 --led-cols=64 --led-chain=2  --led-pixel-mapper=U-mapper 
sleep 2
sudo killall led-image-viewer`

hzeller, 2021-08-17:
why not have the led image viewer do the repeating ? You can pass multiple files to the led image viewer. Use the `-f` option to repeat forever. So something like
```
led-image-viewer -f -w 2 some-image.png -l 1 some-animation.gif
```

soulid, 2021-08-17:
I tried your proposal and it was working well. However I wanted to learn how to use different functions in a bash (animations, graphics, textscroller and so on).
Is that possible anyhow or do I need advanced C coding or Python?

hzeller, 2021-08-17:
I'd probably program that direclty in C++ or Python.

For the shell programs, I'd avoid doing the killing as it might not shut down things properly, but rather use the timeout functionality some of these tools provide, so that they only run for a particular time.

soulid, 2021-08-17:
That'll be fun..never done that...I feel I cannot ask you to do that as an example...you have been such a help creating this software. So I'll try my best...

closed 2021-08-23 by soulid