hzeller/rpi-rgb-led-matrix #1333: Cmake Problem

K0rkunc, 2021-08-21:
Hi my cmake  give error can u help me ? 

[build] [  7%] Building CXX object CMakeFiles/kickass.dir/main.cpp.o
[build] [ 15%] Linking CXX executable kickass
[build] /usr/bin/ld: CMakeFiles/kickass.dir/matrix/lib/framebuffer.cc.o: in function `rgb_matrix::internal::Framebuffer::InitHardwareMapping(char const*)':
**[build] /home/pi/new_kickass/matrix/lib/framebuffer.cc:392: undefined reference to `matrix_hardware_mappings'
[build] collect2: error: ld returned 1 exit status**
[build] make[2]: *** [CMakeFiles/kickass.dir/build.make:249: kickass] Error 1
[build] make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/kickass.dir/all] Error 2
[build] make: *** [Makefile:84: all] Error 2
[build] Build finished with exit code 2

hzeller, 2021-08-21:
It sounds like you need to include hardware-mapping.c in the sources. Maybe you missed it because the other files are *.cc, and this is the only c file.

K0rkunc, 2021-08-21:
> Kaynaklara donanım-mapping.c eklemeniz gerekiyor gibi görünüyor. Belki diğer dosyalar *.cc olduğu için kaçırdınız ve bu tek c dosyası.

Zeller fixed framebuffer but now gives new error 

**[main] Building folder: new_kickass 
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/pi/new_kickass/build --config Debug --target all -j 6 --
[build] [  7%] Building CXX object CMakeFiles/kickass.dir/lib/gpio.cc.o
[build] [ 15%] Building CXX object CMakeFiles/kickass.dir/lib/content-streamer.cc.o
[build] [ 23%] Building CXX object CMakeFiles/kickass.dir/main.cc.o
[build] [ 30%] Building CXX object CMakeFiles/kickass.dir/lib/graphics.cc.o
[build] [ 38%] Building CXX object CMakeFiles/kickass.dir/lib/bdf-font.cc.o
[build] [ 46%] Building CXX object CMakeFiles/kickass.dir/lib/framebuffer.cc.o
[build] [ 53%] Building CXX object CMakeFiles/kickass.dir/lib/led-matrix-c.cc.o
[build] [ 61%] Building CXX object CMakeFiles/kickass.dir/lib/led-matrix.cc.o
[build] [ 69%] Building CXX object CMakeFiles/kickass.dir/lib/multiplex-mappers.cc.o
[build] [ 76%] Building CXX object CMakeFiles/kickass.dir/lib/options-initialize.cc.o
[build] [ 84%] Building CXX object CMakeFiles/kickass.dir/lib/pixel-mapper.cc.o
[build] [ 92%] Building CXX object CMakeFiles/kickass.dir/lib/thread.cc.o
[build] /home/pi/new_kickass/lib/led-matrix.cc: In constructor ‘rgb_matrix::RGBMatrix::Options::Options()’:
[build] <command-line>: error: ‘regular’ was not declared in this scope
[build] /home/pi/new_kickass/lib/led-matrix.cc:270:20: note: in expansion of macro ‘DEFAULT_HARDWARE’
[build]    hardware_mapping(DEFAULT_HARDWARE),
[build]                     ^~~~~~~~~~~~~~~~
[build] make[2]: *** [CMakeFiles/kickass.dir/build.make:154: CMakeFiles/kickass.dir/lib/led-matrix.cc.o] Error 1
[build] make[2]: *** Waiting for unfinished jobs....
[build] make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/kickass.dir/all] Error 2
[build] make: *** [Makefile:84: all] Error 2
[build] Build finished with exit code 2**

**my CMmakelist.txt**
```
cmake_minimum_required(VERSION 3.16.3)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
project(kickass)

set(CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS} -DDEFAULT_HARDWARE="regular")
include_directories(Libs)
include_directories(Include)
include_directories(lib)
include_directories(include)

file(GLOB_RECURSE SRC_FILES Libs/*.cc)
file(GLOB_RECURSE HEADER_FILES Include/*.h)
file(GLOB_RECURSE RGB_SRC_FILES lib/*.cc)
file(GLOB_RECURSE RGB_HEADER_FILES include/*.h)

add_executable(kickass main.cc ${HW_HEADER_FILEs} ${RGB_SRC_FILES} ${RGB_HEADER_FILES} ${SRC_FILES} ${HEADER_FILES})

target_link_libraries(kickass -lpthread curl )
```

hzeller, 2021-08-21:
The word "regular" needs to be a string, so maybe you have to extra-quote it in cmake, maybe with single quotes around? '"regular"'

K0rkunc, 2021-08-21:
i tried single but again same problem

K0rkunc, 2021-08-21:
how to setup standart regular hw map  ?

hzeller, 2021-08-21:
If you give the `cmake` call a `-DCMAKE_VERBOSE_MAKEFILE=true` parameter, it will output the full compiler command line when compiling, so it will be easier to diagnose what is wrong.

Ideally the compiler calls then have a `-DDEFAULT_HARDWARE='"regular"'`, so a single-quoted thing around the double-quoted string. So you essentially you have to find out how to pass a literal string in a cmake CXX_FLAGS.

hzeller, 2021-08-21:
Also, if you don't pass the `DEFAULT_HARDWARE` define, the code should automatically default to `"regular"`. Probably the simplest solution for you, just remove the `-DDEFAULT_HARDWARE="regular"` in your cmake CXX_FLAGS setting.

K0rkunc, 2021-08-21:
yes master, i tried that, now unfortunately i got such an error this time

[build] /usr/bin/ld: CMakeFiles/kickass.dir/lib/framebuffer.cc.o: in function `rgb_matrix::internal::Framebuffer::InitHardwareMapping(char const*)':
[build] /home/pi/new_kickass/lib/framebuffer.cc:392: undefined reference to `matrix_hardware_mappings'
[build] collect2: error: ld returned 1 exit status

K0rkunc, 2021-08-21:
main.cc 

```

#include <iostream>
#include "./include/led-matrix.h"
#include "./include/pixel-mapper.h"
#include "./include/graphics.h"
#include <time.h>
#include <sys/types.h>
#include <unistd.h>
#include <algorithm>
#include <map>
#include <string>
#include <limits.h>
#include "./include/gpio.h"
#include <unistd.h>
#include <math.h>
#include <stdio.h>
#include <signal.h>

using rgb_matrix::RGBMatrix;
using rgb_matrix::Canvas;

volatile bool interrupt_received = false;
static void InterruptHandler(int signo) {
  interrupt_received = true;
}

static void DrawOnCanvas(Canvas *canvas) {
  /*
   * Let's create a simple animation. We use the canvas to draw
   * pixels. We wait between each step to have a slower animation.
   */
  canvas->Fill(0, 0, 255);

  int center_x = canvas->width() / 2;
  int center_y = canvas->height() / 2;
  float radius_max = canvas->width() / 2;
  float angle_step = 1.0 / 360;
  for (float a = 0, r = 0; r < radius_max; a += angle_step, r += angle_step) {
    if (interrupt_received)
      return;
    float dot_x = cos(a * 2 * M_PI) * r;
    float dot_y = sin(a * 2 * M_PI) * r;
    canvas->SetPixel(center_x + dot_x, center_y + dot_y,
                     255, 0, 0);
    usleep(1 * 1000);  // wait a little to slow down things.
  }
}

int main(int argc, char *argv[]) {
  RGBMatrix::Options defaults;
  defaults.hardware_mapping ="regular"; 
  defaults.rows = 32;
  defaults.chain_length = 1;
  defaults.parallel = 1;
  defaults.show_refresh_rate = true;
  Canvas *canvas = RGBMatrix::CreateFromFlags(&argc, &argv, &defaults);
  if (canvas == NULL)
    return 1;

  // It is always good to set up a signal handler to cleanly exit when we
  // receive a CTRL-C for instance. The DrawOnCanvas() routine is looking
  // for that.
  signal(SIGTERM, InterruptHandler);
  signal(SIGINT, InterruptHandler);

  DrawOnCanvas(canvas);    // Using the canvas.

  // Animation finished. Shut down the RGB matrix.
  canvas->Clear();
  delete canvas;

  return 0;
}

```

hzeller, 2021-08-21:
The `matrix_hardware_mappings` is defined in the C-file as [mentioned above](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1333#issuecomment-903128390) . So you have to add that in your cmake file
```
file(GLOB_RECURSE RGB_SRC_FILES lib/*.cc lib/*.c)
```

K0rkunc, 2021-08-21:
> `matrix_hardware_mappings`C-dosyasında tanımlanır [yukarıda bahsedilen](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1333#issuecomment-903128390) . Yani bunu cmake dosyanıza eklemelisiniz
> 
> ```
> file(GLOB_RECURSE RGB_SRC_FILES lib/*.cc lib/*.c)
> ```

sir you are amazing!!

lib and thank you very much for your help.

I've included it below if anyone wants to use it.
CMAKE EXAMPLE 

cmake_minimum_required(VERSION 3.16.3)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
project(kickass)

include_directories(Libs)
include_directories(Include)
include_directories(lib)
include_directories(include)

file(GLOB_RECURSE SRC_FILES Libs/*.cc)
file(GLOB_RECURSE HEADER_FILES Include/*.h)
file(GLOB_RECURSE RGB_SRC_FILES lib/*.cc lib/*.c)
file(GLOB_RECURSE RGB_HEADER_FILES include/*.h)

add_executable(kickass main.cc ${HW_HEADER_FILEs} ${RGB_SRC_FILES} ${RGB_HEADER_FILES} ${SRC_FILES} ${HEADER_FILES})

target_link_libraries(kickass -lpthread )

closed 2021-08-22 by K0rkunc