hzeller/rpi-rgb-led-matrix #1334: Loop stops after a few hours

soulid, 2021-08-24:
Not sure if this is a issue to the script. I do start the the script with followin code. After a few hours the system stops...and I do not know why:

`sudo rpi-rgb-led-matrix/utils/video-viewer -F  -f rpi-rgb-led-matrix/utils/Versuch1710.webm --led-rows=32 --led-cols=64 --led-chain=2  --led-pixel-mapper=U-mapper;Rotate=180`

May someone is more knowledgable than me helping...

hzeller, 2021-08-29:
another idea: did you start this on a remote shell and that got closed for whatever network reason at some point, so that it took the process with it ? In that case, maybe adding the `--led-daemon` will help as it puts the process in the background.

soulid, 2021-08-29:
Got the issue. I started the script by SSH and as soon as I log off the SSH the script stops working.
I solved my issue to add this to the .bashrc. Now it works perfectly fine- I cannot cancel the sript but found a workaround by renaming the movie source directory and therefore cancel the executetion)

soulid, 2021-08-29:
I will try to use the --led-daemon. Thynks for your help!

hzeller, 2021-08-29:
In your shell, you can stop the process by calling 
```
sudo killall video-viewer
```