hzeller/rpi-rgb-led-matrix #1337: Columns Randomly Flickering

drewblank, 2021-09-07:
I'm getting random columns flickering on my 64x64 led matrix.  
Video showing issue: https://photos.app.goo.gl/FtsynjVkL16yp2xV9

Using pi zero, adafruit bonnet, and HUB75 64x64 led matrix. Followed install guide on Adafruit's website. 

This random column flickering is not unique to the mlb score program, happens with any of the demos. Installed 'quality' mode, soldering GPIO 4 to 18. Tried different --led-slowdown-gpio values with no discernable difference. Changing brightness doesn't change anything. 

One thing I noticed is the power supply output is a little high, rated for 5V but outputting 5.3V...not sure if that would cause the column flickering.

hzeller, 2021-09-07:
And you also use `--led-gpio-mapping=adafruit-hat-pwm` I suppose ? These glitches look like if the hardware PWM was not used, so just double-checking.

Tha Pi Zero is a little bit challenged, as it is essentially the 2012 CPU with one core, I could imagine that it can have some brightness fluctuations, but the single-line glitches looks more like not using the hardware pwm.

drewblank, 2021-09-07:
I'm not quite sure how I missed that flag in all my trouble shooting, but then again I confused rows and columns so...

But that seemed to fix it right away, thank you!

hzeller, 2021-09-08:
Alright, have fun with your LED panel!

closed 2021-09-08 by hzeller