hzeller/rpi-rgb-led-matrix #1338: I have a Issue with displaying a Gif or a Image.

OskarGER05, 2021-09-15:
The Gif or Image is doubled in diffrent colors than the original one and it is cut in half.
[image]
I run the command: sudo ./led-image-viewer Pikachu.gif
it said: Size: 32x32, Hardware gpio mapping: regular
I am using a Joy-It 64x64 LED-Matrix panel

Sorry if my english isn't that good.

OskarGER05, 2021-09-15:
The Demo works completly fine... i am a bit confused

hzeller, 2021-09-15:
You have a 64x64 panel, so make sure to supply `--led-rows=64 --led-cols=64` flags.

Since there are many types of displays, you have to go through the [types of display](https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays) section in the readme to experiment if you also have to need some multiplexing option.

OskarGER05, 2021-09-15:
ok thank you i will try that out

closed 2021-09-19 by OskarGER05