hzeller/rpi-rgb-led-matrix #1339: MATRIX SCRAMBLES AFTER 3 SECONDS

perrySeoul, 2021-09-18:
Good morning. Thank you, I really appreciate your work. I have been using the rgb-led-matrix clock for about two years. The SD card became corrupted after updating raspbian last week. I did not write down or save the settings I was using for the clock. I am using a Raspberry Pi Zero W. I put raspbian on a new SD card, downloaded the rgb-led-matrix. When I run the clock example or any example the matrix is stable for 3 seconds then scrambles.

This is the settings I am using now.

`sudo ./clock -f ../fonts/7x13.bdf --led-chain=2 --led-rgb-sequence=BGR -d "%A" -d "%H:%M:%S"'

Here are pictures of the back of the matrix, and the issue.
[image]

[image]

[image]

[image]

[image]

Thank you for your help.
Perry

hzeller, 2021-09-18:
Did you check the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) ?
Sudden scrambling is consitent with what one would observe if one-wire protocol is enabled.

perrySeoul, 2021-09-18:
Thank you for your help. I’m sorry for wasting your time with this issue.
This is why I hesitated to ask for help. I knew this was something I should
have resolved on my own. I did many google searches, I did read the
troubleshooting guide.

I missed the solution to resolve the issue.

Thank you for your time and help.

Perry

hzeller, 2021-09-18:
cool, sounds like it is working now. Have fun with your LED panel.

closed 2021-09-18 by hzeller