hzeller/rpi-rgb-led-matrix #1340: Minor flickering issue on 128x64 Matrix (w/ Adafruit RGB Matrix Bonnet)

mobeigi, 2021-09-18:
Hi all,

I have recently completed a project to build out a 128x64 matrix. It is working well but has a slight flickering issue. The flickering is basically every 10 seconds or so a half row of pixels or some pixels near each other go white or dim themselves. This is a momentary flicker. I can probably live with it but it would be nice to fix.

Here is the hardware setup I have:
[image]

My matrix: https://www.aliexpress.com/item/32980828995.html

As you can tell power supply is plentiful and wire quality is high (14 AWG for the DC power for the Adafruit which feeds the LED).

I get this flickering in two scenarios:

Demo 4:
`./demo --led-rows=64 --led-cols=128 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 --led-panel-type=FM6126A --led-slowdown-gpio=4 -D 0`

Image viewer with a full screen image:
`python image-viewer.py my-picture.jpg`

I do not get any flickering at all if using some other demos (mainly demos that use < 90% of pixels at once).

Things I have tried with no difference at all in the flickering:
- Use `isolcpu 3` in boot and reboot (My RPI 4 has 4 cores, core 4 maxes out to 57% max during demo 4)
- Tweak `DFIXED_FRAME_MICROSECONDS` in make file
- Try to `limit_refresh_rate` (I get roughly 89hz by normal but reducing this doesn't make flicker go away)
- Change `pwm_bits`, `pwm_lsb_nanoseconds` and `pwm_dither_bits`
- Lots of other things in combination with each other

I'm not sure if the hardware flickering mod will be applicable to me? So far I had to solder the 8 on the Adafruit RGB Matrix bonnet to ground to support the E line for my matrix but have made no other modifications.
[image]

Thanks for any help, I really tried to use existing resources here to solve the issue myself but am a little lost! Awesome work on this repo and documentation though, I love it!

hzeller, 2021-09-18:
Yes, you need to do the [PWM hardware fix](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) for the Adafruit board. The bonnet is slightly different than the HAT, but essentially you need to do the equivalent wiring.
That is an absolute necessity to make sure you have stable timings with no glitchy single lines. I suspect that is what is happening here.

There could be other reasons, but only after you have solved the PWM hack, they are worth looking at. Checking all things in the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting). is probably a good idea. E.g. regular flicker can happen if 1-wire protocol is enabled,

The power supply or long cable length might not be able to deal with the transients easily, so it might help to add low-ESR capacitors right at the connectors of the panel.

mobeigi, 2021-09-18:
Wonderful @hzeller !

I just added the jump wire solder from 4 to 18 (this guide also mentions it for my exact Adafruit product: https://cdn-learn.adafruit.com/downloads/pdf/adafruit-rgb-matrix-bonnet-for-raspberry-pi.pdf).

I then disabled onboard sound on my Raspberry Pi using README steps and added `--led-gpio-mapping=adafruit-hat-pwm` to my flags.

It now runs demo 4 and the image viewer perfectly with no flicker what so ever! Now its truly perfect, thanks again 👍

closed 2021-09-18 by mobeigi