hzeller/rpi-rgb-led-matrix #1341: Mapping issue (glitching fixed)

shotbyapony, 2021-09-18:
https://imgur.com/a/mt6GlHH

--led-no-hardware-pulse  fixed the flickering seen, but I am still having the issue of it only using half the panel. I am using `sudo examples-api-use/demo -D4 --led-rows=64 --led-cols=64 --led-slowdown-gpio=4 --led-show-refresh --led-chain=3 --led-parallel=3 --led-no-hardware-pulse` . There are 2 in daisychain, 3 chain in parallel to an Electrodragon, and they are 64x64 panels. Thank you for any help, not really sure where to go from here.

hzeller, 2021-09-18:
You definitely _want_ the hardware pulse, because it reduces flickering. I can't think of a situation in which no-hardware-pulse would be better. Could you elaborate how it manifests with the different settings ?

For the 64x64 panel, make sure to connect also the E line, which may be on either pin 4 or 8 on your panel. It would explain why half of the panel is not lit.

Out of curiosity: if this is a cube, you'd have 6 panels. But with led chain of 3 and led parallel of 3 you are addressing 9 panels. Is that intentional or just a glitch in the command line above ?

shotbyapony, 2021-09-18:
Hi there thank you for the ,

hzeller, 2021-09-18:
it is hard to tell what the flickering is from the image, but also make sure to disable 1-wire communication if that is enabled (see troubleshooting section)

shotbyapony, 2021-09-18:
Sorry about that, the virtual keyboard on my laptop entered the next plane of existence and closed the issue. I am following this guide and wired it like him, the demo code is on the pdf, and I am not sure what the E line is, it is a panel with 4power in pins and 16 data pins one in one out, he explains why they are 3x3 in the pdf, and I am not near the project right now. I will try to disable one wire later this evening, thank you for the help . The hardware pulse fixed all the flickering but maybe it is coincidence, i will try to tweak with it later.

hzeller, 2021-09-18:
In doubt follow the guide in this library not of what is found somewhere else.  The [wiring guide](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#wiring-diagram) lists what the E-line to connect. Depending on your panel, that is either on Pin 4 or pin 8 on the panel connector.

What is the guide you are referring to ?

shotbyapony, 2021-09-18:
Oh dear, I must be losing my marbles over here, the link to the guide is https://drive.google.com/file/d/1w1jja1O2fJ9RGRqpyONtwTtHGkaTaXRy/view?usp=sharing , once again thank you for the help!

hzeller, 2021-09-18:
If you use the electrodragon board, make sure to enable the E line there.
I think the jumpers are similar to the [adapter in this project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing)

shotbyapony, 2021-09-18:
I’ll look in to it later, thank you! Will let you know how it goes

shotbyapony, 2021-09-19:
Banding and flickering fixed without the pulse turned off, and I jumped 2 pins on the dragon to enable E line. Now the banding is gone, but the mapping is all off, how do I fix this?

hzeller, 2021-09-19:
what do you mean with 'mapping is off' ?

closed 2022-01-06 by masonbesmer