hzeller/rpi-rgb-led-matrix #1342: ICND2055 not working

fecri, 2021-09-24:
Hello everyone,

I have a problem with Rapsberry Pi and a p2.5 panel 128*64 (e-line).
The chipsets are ICND2055.
The screen is not light up.

Can you help me to implement a new logic for this kind of chipset ?

The datasheet: [ICND2055_datasheet_EN_2021_V1.8b.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/7224362/ICND2055_datasheet_EN_2021_V1.8b.pdf)

Thank you.

hzeller, 2021-09-25:
How did you connect it ? With or without level shifters or do you use one of the existing [adapters](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) ?

fecri, 2021-09-25:
> How did you connect it ? With or without level shifters or do you use one of the existing [adapters](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) ?

Hi, i use active-3 adapter from https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/.

Thank you

hzeller, 2021-09-25:
Ok, so what did you try to make it work, what command line options are you using and were testing ?

fecri, 2021-09-25:
I tried commands likes above:

./led-image-viewer --led-cols=128 --led-rows=64 --led-row-addr-type=4 --led-slowdown-gpio=3 --led-panel-type=FM6126A -f ...

./led-image-viewer --led-cols=128 --led-rows=64 --led-row-addr-type=4 -f ...

hzeller, 2021-09-25:
The datasheet of the ICND2055 is confusing and one would probably have to probe directly with an oscilloscope to better understand how it is integrated in the led matrix panel. The interesting part is that it seems to have a very long shift register for all rows ?
I don't have a ICND2055 panel to test, so you'd need to do that work and write up who it works there and ideally send a pull request to support.

fecri, 2021-09-25:
I will work on this. thank for all :)

fecri, 2022-01-03:
Hi sorry for the late response. I don't have solution for this issue.

fecri, 2022-01-03:
> The datasheet of the ICND2055 is confusing and one would probably have to probe directly with an oscilloscope to better understand how it is integrated in the led matrix panel. The interesting part [...]

Can I ship you a panel ? I have 10 panels, you can keep it.
I will pay shipment fees.

twessel, 2022-10-27:
Has anybody ever found a solution for this? I'm running into the same problem and cannot figure out how to get the panels to work. 

>  to probe directly with an oscilloscope to better understand 

I have an old oscilloscope and would give it a try, if you could give me a hint how to attack this. Would your demo program be useful to run and probe against?

closed 2025-11-27 by marcmerlin