hzeller/rpi-rgb-led-matrix #1343: I am trying to display the album cover of a song

OskarGER05, 2021-09-25:
Hi !
I followed this [reddit post](https://www.reddit.com/r/raspberry_pi/comments/ombwwg/my_64x64_rgb_led_matrix_album_art_display_pi_3b/?utm_medium=android_app&utm_source=share)

I have this rgb matrix: [Joy-It-64x64](https://joy-it.net/de/products/LED-Matrix01)

I run this command on the pi: ```sudo ./flaschen-taschen/server/ft-server --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=2 --led-rows=64 --led-cols=64 --led-show-refresh --led-brightness=50```

I didnt get it to work, i dont know if you need more information to help me.

hzeller, 2021-09-25:
I'd first ask you to read the full README on https://github.com/hzeller/rpi-rgb-led-matrix - in particular the 64 columns might need the manual breaktout of the E line, otherwise there will be two black stripes.

Before  running the flaschen-taschen server, maybe first start seing if the [demo](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use) is working. Once these work, I'd switch to the FlaschenTaschen server.

Also when sending stuff to the flaschen-taschen server with the clients https://github.com/hzeller/flaschen-taschen/tree/master/client make sure to set the hostname or IP address of your matrix with the `-h` option, otherwise the images are sent into the void :)

OskarGER05, 2021-09-25:
i checked the things you named above, but it still isnt working. Any ideas what i could do now?

hzeller, 2021-09-25:
What are the things you tested to get it working ? You need to give some details otherwise it is hard to help - what is your power supply ? Did you measure that power is there, which services did you disable, what is the Pi you're running it on etc.

OskarGER05, 2021-09-25:
I am using a LPV-60-5 Powersuply with 1.2 A 5V. I tested a gif and a picture. Power is 100 % there. On the pi i just have VNC running.
I think that it is a problem with the connection via the flaschen-taschen server from my pi and my pc. I checked the rainmeter scripts but didnt find anything.

hzeller, 2021-09-25:
1.2A is a little bit on the low side, you might want to look for 5A or so for one panel. With all black pixels, that should be fine :)

So when you suspect that it is the flaschen taschen connection, then you tried the demos and they all work ?

OskarGER05, 2021-09-25:
Oh sorry the powersupply has a 8 A and 5v output

OskarGER05, 2021-09-25:
I didnt get the demos to work

OskarGER05, 2021-09-25:
I just need this https://github.com/kylejohnsonkj/ft-artwork/blob/master/UpdateArt.ps1
to send something to the server? Am I right?

hzeller, 2021-09-25:
Ok, try get the demos to work first, then we can worry about the FlaschenTaschen situation. If the demos are not working, the flaschen-taschen server will also not be able to display anything.

Systematically check the options relevant to panel selection such as `--led-row-addr-type` and `--led-panel-type`. Since you connected it with the Adafruit hat, also don't forget to add that gpio mapping in all these experiments, because the wrong mapping there will of course not light up anything and then you're chasing ghosts while playing with the other options.

OskarGER05, 2021-09-25:
The demos did work.

hzeller, 2021-09-25:
Ah, because before you said the demos didn't work. Alright so if you have your matrix display stuff with the demos, then you can move on to the flaschen-taschen, and just give it all the same `--led-...` options you have used for the demos.

The above script will work if you have your PI on IP address `192.168.1.44` and have compiled your [send-image](https://github.com/hzeller/flaschen-taschen/tree/master/client#send-image) utility in the same directory the script is running. Otherwise, I'd probably suggest to use an absolute path to the send-image utility.

I'd try to `send-image` utility first manually before trying it in the script of course. Once you get an image sent with that, use the workign parameters in the script.

OskarGER05, 2021-09-26:
Ok thank you

closed 2021-09-26 by OskarGER05