hzeller/rpi-rgb-led-matrix #1358: How to create and read stream file?

K0rkunc, 2021-11-02:
hello, I have examined your image viewer example, but I could not understand it because it is too complicated..

please don't tell me
How to create the stream output of the gif file and save it in the directory I want.
Can you describe how to read the -stream file and copy it to the canvas?

hzeller, 2021-11-02:
The `include/content-streamer.h` has everything you need.

First, you need to create a StreamIO of your liking, if you read from a file, that will be a `FileStreamIO`. Then you can use that to get the frames with a `StreamReader`

So roughly something like this:

```C++
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include "content-streamer.h"

int main() {
   // initialize matrix ...
   FrameCanvas *offscreen_canvas = matrix->CreateFrameCanvas();

  const int fd = open("/path/to/file", O_RDONLY);
  rgb_matrix::FileStreamIO io(fd);
  rgb_matrix::StreamReader reader(&io);
  uint32_t wait_time;
  for (;;) {
    while (reader.getNext(offscreen_canvas,  &wait_time)) {
      offscreen_canvas = matrix->SwapOnVSync(offscreen_canvas(offscreen_canvas);
      usleep(wait_time);
    }
    reader.Rewind();
  }
}
```

K0rkunc, 2021-11-02:
[image]
getting error

K0rkunc, 2021-11-02:
Why no member getnext

hzeller, 2021-11-02:
It is upper-case: `GetNext()` https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/content-streamer.h#L92

K0rkunc, 2021-11-13:
> It is upper-case: `GetNext()` https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/content-streamer.h#L92

i tried but panel dont lights but program works

K0rkunc, 2021-11-15:
solved im creat animation file and work  how to get frame num ?

K0rkunc, 2021-11-15:
and rgb matrix seettings not work

hzeller, 2021-11-15:
It is a stream, you don't know the number of frames beforehand, but you'll know when the stream ends, because `GetNext()` returns false.
The second message is cryptic, it is not clear what you mean. Remember, if you want someone to help you, you need to provide all relevant information (set-up, settings, command line flags, expectation, observed result instead...).

K0rkunc, 2021-11-15:
my masters
We make rgb matrix settings at the start of main options.brightness options.cols... etc..

these settings I defined are not valid in stream file. Stream file opens with its own settings. How can I dictate my own settings to it?

Thank you very much for your time, I will give you a little more exposure.

actually my main problem is my program is in the following additional file, if you have time, please check it out.

I created 2 threads with one of these threads, I print coalescend image files on the canvas and play them.

With the second thread, I convert the scene transition from the picture file to a collage while the previous scene is playing, in order not to waste time.

but during this translation, there are tiny flickers on the screen. How do you think I can solve this problem?

Since my English is not good, I used translate, sorry for the word mistakes.

mycodes
https://www.codepile.net/pile/pj00pmjn

K0rkunc, 2021-11-18:
@hzeller  When I use file stream it doesn't apply runtime options and matrix options. How can I fix this?

closed 2025-11-27 by marcmerlin