hzeller/rpi-rgb-led-matrix #1361: Changing example code

dunlaveys, 2021-11-05:
I know all of this is because of my inexperience with the matrix, C & C++, and makefiles, but I'm trying to learn.

I want to take the c-example and play around with some of the code to get a feel for how the matrix behaves. 
Let me first check that it is running as expected: The LEDs start with a slight blue/green tint, and cycle up to bright red as the loops run. It does that a few times and then exits. Next, let's say I want to start REALLY simple, so I just switch around the RGB values in the set_pixel parameters (line 42). Should be an easy way to begin, but I immediately hit troubles with compiling/making which is where my inexperience with multiple libraries and all that run into a brick wall that I can't seem to get around.

From the beginning:
I install everything from github, following the tutorial on adafruit. I do the curl command followed by the "bash rgb-matrix.sh". I select the hat option, then convenience (multipurpse pi, just playing around, maybe switch when I have an actual project in mind). Everything works perfectly. I've figured out the options to match my matrix size (64x64) and slowdown the gpio since I'm on a pi4. Time to dig in and figure out what the code does.

I edit the .c file and try to compile. I ran into some early trouble because the includes are in a different folder, but I quickly figure that out. I try following the section in the makefile for the commands, but cannot get the compilers to generate a new executable. I *know* it is because I'm a novice and just used to compiling a c or c++ program where everything is in one folder, but now I'm trying to compile something that uses a bit of both. To test, I make a copy and call it c-examples2.c. I get the .o but cannot get the binary to be created. 

Ok, the makefile has all of this sorted out. I figure if I just do a "make" in the examples-api-use directory, it'll handle the recompile for me. As soon as that finishes, I get an error about the snd_bcm2835: found that the Pi sound module is loaded, etc etc, exiting out. I assume this is related to the "quality" vs "convenience" option from install and now the code thinks it is in the other mode. But it was just working when the original install made all the binaries. Trying to run it with no-hardware-pulse doesn't result in any error. It sits and stares at me long enough for me to think the code is running, but no lights. Only way I've found to fix it is to reinstall from github and overwrite everything.

After a couple go-arounds with that, I got smart enough to make my own directory for my examples. I copied c-example and the makefile over and edited the makefile to get rid of all the references to the  demo and other examples. Makes without errors, but same results as before. At least now I can just delete this directory and recopy from the original files without a fresh install.

I have just enough knowledge to know it is probably something stupidly simple that I just don't know, but it has stopped me dead in my tracks just as I was going to begin experimenting. It is also something that I don't know well enough to phrase correctly on Google or StackOverflow to get the right results.

Any help would be appreciated. Thank you.

greatballoflazers, 2021-11-06:
Pi 4 needs max slow down. Two Options.
1) Run with --led-gpio-slowdown=5
2) Change code use runtime options

After line 13:
``` C
struct RGBLedRuntimeOptions runtime_options;
```
After line 21:
``` C
memset(&runtime_options, 0, sizeof(runtime_options));
runtime_otions.gpio_slowdown = 5;
```
Change line 23 and 24 to this:
``` C
matrix = led_matrix_create_from_options_and_rt_option(&options, &runtime_options);
```

This would in the blind spot. Honestly what I could do is mess around with demo -D 0 to get settings right. Then run c example using those settings make sure the library is happy. By default you can overwrite with command line. Then from there you can change code once you have the timings and stuff settled.

dunlaveys, 2021-11-06:
I'm sorry if I was unclear in my original post, if my post was TLDR for you, or if I'm missing the connection, but it looks like you're telling me to fix a problem I've already taken care of by the method I can't get to work.

As I said in my second paragraph, I've already figured out the slowdown. When I run the demos, I flag slowdown 5 and everything works perfectly.

But then you say to change the code... that's fine and good, but my issue is that I can't get code to recompile and still run. If I just call the c compiler and/or the c++ compiler, no new executable is created (the original remains unchanged). If I try the "make" command, something changes in the binaries and none of the demos or examples will run.

As simple as I can phrase the situation:
Immediately after install: sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=5
Everything is perfect.

Type 'make', press enter, and get prompt: sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=5
Get an error: "snd_bcm2835: found that the Pi sound module is loaded." Etc
Try to run with --led-no-hardware-pulse and get nothing

So I can't just change the code. That is what I am asking for help on.

DavidSzczecina, 2022-01-07:
@dunlaveys I just ran into the problem you described and I'm stuck in the same place. Any chance you figured out what the solution was? Thanks

dunlaveys, 2022-01-07:
Yes, I did. Sorry, I always hate it when the OP solves the problem, but
doesn't put the solution on their thread and leaves everybody else hanging.
Then I go and do the exact same thing.

Anyway, the issue is that the examples-api-use makefile includes a call
over to the /lib makefile. The options you selected when you installed
aren't saved anywhere and get overwritten. As soon as you compile your new
code with 'make', the library gets rebuilt for the default hardware
settings. Go into /lib/Makefile with your text editor, and change all the
defaults to your hardware options. The main one for this issue is on line
37, change the default from HARDWARE_DESC?=regular to whatever you're using
(adafruit-hat-pwm in my case).

Like I said in my original post, it was my own inexperience that was
holding me up. I finally freed up time to wade through all the lines in the
makefile and trace them back to where those settings were being overwritten.

closed 2025-11-27 by marcmerlin