hzeller/rpi-rgb-led-matrix #1364: 64x32 Panel don't work

henri-ngl, 2021-11-16:
Hi!

First of all thank you again for all your work. It's amazing. 

So i just bought some 64x32 Panel 
[image]
that i'm an raspberry pi 3 and an electrodragon hat 
[image]
; all powered with 5V-60A power supply.

The problem is that nothing shows on the screen. When i use my old 64x32 every thing is OK but those new screen won't work. i tried the example of your library with different flags but still won't work. the screen remain off
[image]

It have the identical information as the old one  but none of them work. It's there a way to figure it?

Thank you

greatballoflazers, 2021-11-16:
According to [this](https://lucasteske.dev/lyd6168) it is not supported. It is believed to be a rebranded MBI5153 or ICN2053. (This has been reported multiple times.) This is a PWM based panel, which is not supported by this project.

It is possible to use receiver card, like the one mentioned in the post, directly via NetCard. There are a few people which have gotten this to work. There is some code out there also, however it was reported by one of them that there could be a stability issue in some cases for the PWM panels. There is a work around, kind of. Sender cards also work well. If yours works as ICN2053 there is some code out there for it on the ESP32. 

No one has really sat down and overhauled this code base to support PWM based panels for the code base. A long time ago someone mentioned it would be possible. However most of their posts were removed. These store PWM bits differently than non-PWM panels do. They also have a different state machine for multiplexing.

These panels appear in the higher density, lower Px panels. They tend to be a little more expensive. Some of these have fancy features too. They are nice for larger displays but are more expensive for smaller displays. There are some non-PWM panels at lower density but getting higher quality is a little harder here. Receiver card can do better than Pi due to connector count, however that depends if it matters which is a topic in of itself.

Lot of the panels are modular. The ICs are usually fairly interchangeable. They make runs of different configurations. Sometimes it is helpful to ask which configuration they have in which density.

DerekRead, 2023-02-06:
I have the same panel and can almost get it to display scrolling text. "hello world!" is almost legible but rows are missing.

I've tried a few different identical panels to try to eliminate the panel as the issue, they all suffer the exact same missing lines so most likely this is software not hardware.

I believe the missing lines are likely because this panel is probably 1:16 scan (?) which does not seem to be supported by this project (I have not found a setting for this).

I am running this from an rpi 4b, so found I had to use the param "--led-slowdown-gpio".
Brightness is set to 25 to reduce any possibility that this is an amp issue, even though I have a 4A supply which should be more than enough.

Reading the docs it looks like I should be using --led-panel-type=4 (for ABCD) but that appears much worse than when I have it set to 2.

These are the best settings I've found so far (also note I'm using the adafruit hat):
`python3 runtext.py --led-rows=32 --led-cols=64 --led-multiplexing=1 --led-row-addr-type=2 --led-gpio-mapping=adafruit-hat --led-brightness=25 --led-slowdown-gpio=4
`
[image]

DerekRead, 2023-02-06:
I see now that characters are also individually mirrored and maybe flipped. More clues. Hmmm.

DerekRead, 2023-02-06:
@henri-ngl: I guess this may not help you if your panel is completely blank but I thought I'd post just in case, given that I have the identical panel (from your rear photo), and I have seen settings with this software where the panel will appear blank.

tldr: using an rpi4b and the adafruit hat all I need is this (yes, I see you are using electrodragon):
`runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4`

I was initially thrown off because I didn't use --led-slowdown-gpio=4
and was trying all other variations of --led-panel-type & --led-multiplexing & --led-row-addr-type I found these were all slightly or badly messed up (some were close) and in a few cases the screen **would be blank** <-- which sounds like your issue,  maybe?

Maybe this is useless info for you but maybe not.

I also pulled the current version of this matrix code before doing my last tests. There were quite a lot of changes since I last tested. That might have altered the behaviour as well.

[image]

anandrajgupta, 2023-02-06:
--led-multiplexing=0

davemaster, 2023-02-06:
> --led-multiplexing=0

I think that value is the default. Try with 0-14 (top I think) values with ONE PANEL (power supply?)

DerekRead, 2023-02-06:
Yes, 0 is the default. Thanks. For me it's working great now (single panel for this project). I had seen other people with other settings and that threw me way off track testing a bunch of settings in combinations, which wasn't necessary at all.

So, for this panel (see pic from OP but using adafruit hat) all I need are these settings for single panel: `runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4`

closed 2025-11-27 by marcmerlin