hzeller/rpi-rgb-led-matrix #1365: Matrix started to get dimmer for 10-15 seconds every few minutes

Boerny41, 2021-11-16:
Hey everyone,

overall the matrix is working just fine however after a few month if not a year by now, it started to get dimmer and then brighten back up after 10-15seconds. This happends every few minutes.

I'm more or less sure that it is a power supply issue (5v 10a laptop style) because i tried another one with same specs and it stopped there. However with the other power supply there are random white dots showing up. e.g. one led flashing white than another and another multiple at a time. Besides that it does a horrible noise... 

I do not want to buy tons of power supplies until one works and keeps working so I though maybe there is a way to solve it with software.

It is a 64x64 matrix (2x 32x64 panel) run by a pi4 with the cheap header from ElectroDragon. Brightness is set to 80% however the power supply also manages 100% I just like it a bit dimmer. Refresh rate is set to 200mhz. If you need any more information please let me know

hzeller, 2021-11-16:
If you suspect it a power supply issue, make sure to keep short and sufficiently thick power cables and possibly add low ESR capacitors directly at the power input of the matrix as described in the [power wiring section](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power). Noise could indicate that you're hitting a frequency that the switching power supply has trouble compensating, so adding the capacitor might well do the trick.

Boerny41, 2021-11-17:
Thanks for the quick answer. No idea how I missed that but surely going to try it with capacitors.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin