hzeller/rpi-rgb-led-matrix #1366: Update message from TCP server

forever80-dev, 2021-11-17:
hello, i write this code but message cleared after time sleep two second. 
I would like it to remain displayed until a new request but I cannot find the solution. If anyone has a lead thank you

#!/usr/bin/env python
# Display a runtext with double-buffering
import argparse
from rgbmatrix import RGBMatrix, RGBMatrixOptions
from rgbmatrix import graphics
import time
import socket
import thread
import urlparse
from urlparse import parse_qs
import BaseHTTPServer

class handler(BaseHTTPServer.BaseHTTPRequestHandler):
     def do_HEAD(s):
         s.send_response(200)
         s.send_header("Content-type", "text/html")
         s.end_headers()
     def do_GET(s):
         parsed_path = urlparse.urlparse(s.path)
         s.send_response(200)
         s.send_header("Content-type", "text/html")
         s.end_headers()
         s.wfile.write("OK"+parse_qs(parsed_path.query)['scoreB'][0])
         scoreA=parse_qs(parsed_path.query)['scoreA'][0]
	 scoreB=parse_qs(parsed_path.query)['scoreB'][0]
         SendMatrix(scoreA,scoreB)

def SendMatrix(scoreA,scoreB):
    print('Matrix')
    font = graphics.Font()
    red = graphics.Color(255, 0, 0)
    blue = graphics.Color(0, 0, 255)
    options = RGBMatrixOptions()
    options.rows = 32
    options.cols = 64
    options.chain_length = 3
    options.parallel = 2
    options.multiplexing = 1
    matrix = RGBMatrix(options = options)
    offscreen_canvas = matrix.CreateFrameCanvas()

    if len(scoreA)>2 or len(scoreB)>2:
        fontchoise="pygScore3.bdf"
        posx=50
    else:
       fontchoise="pygScore2.bdf"
       posx=60
    font.LoadFont("../../../fonts/"+fontchoise)

    offscreen_canvas.Clear()
    lenA=graphics.DrawText(offscreen_canvas, font, 2, posx, red, scoreA)
    lenB=graphics.DrawText(offscreen_canvas, font, 100, posx, blue, scoreB)
    offscreen_canvas.Clear()
    graphics.DrawText(offscreen_canvas, font, (100-lenA)/2, posx, red, scoreA)
    graphics.DrawText(offscreen_canvas, font, ((100-lenB)/2)+100, posx, blue, scoreB)
    offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)
    time.sleep(2)

if __name__ == '__main__':
     server_class = BaseHTTPServer.HTTPServer
     httpd = server_class(('', 5000), handler)
     try:
         httpd.serve_forever()
     except KeyboardInterrupt:
         pass
     httpd.server_close()

forever80-dev, 2021-11-18:
it seems this happens when there is a write to memory

Beersatron, 2021-11-18:
I have seen flicker when the ethernet port was in use, so I switched to WiFi only and the flicker was gone. Are you using ethernet?

forever80-dev, 2021-11-18:
Yes I am using ethernet. I will test in WIFI

forever80-dev, 2021-11-19:
The problem is the same. And when I read video. it is really more pronounced.

Beersatron, 2021-11-19:
What is your setup? What rPi, how many panels and are you using a HAT or just wired up? How much power are you supplying?

What do the demos look like and can you try the demos without using Python?

Beersatron, 2021-11-19:
Also, I believe the best setup is to use a lite version of the Raspbian OS and run things headless.

forever80-dev, 2021-11-19:
I have two panel 3x64, power supply is 40A 5v

I have Raspberry pi 3 b+ and the card : DEBO MATRIXCTRL Raspberry Pi - Ecran pour matrice LED RGB

Command is : video-viewer -f --led-rows=32 --led-cols=64 --led-parallel=2 --led-chain=1 --led-multiplexing=1 logo.mp4 -F -f

closed 2025-11-27 by marcmerlin