hzeller/rpi-rgb-led-matrix #1367: Problems installing with Python 3.9

TerryS719, 2021-11-18:
I am experiencing errors when installing using the script on the Adafruit matrix bonnet pages.  Searching on the errors leads me to believe it is due to changes in Python 3.9.   I previously had been using Python 3.7 without problems, but now I am using a new Pi Zero 2 W so am using the latest OS which has moved to Python 3.9
 There are several errors, but this is the first:
error: ‘pytypeobject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
It looks like 3.9 drops tp_print
Is anyone else seeing this issue?

HFabi, 2021-11-18:
Hey. I am having the same issue and currently no idea how to fix it.

TerryS719, 2021-11-18:
I tried version 3.8 and had the same issue, but then tried version 3.7.11 and RGB Matrix installed.  The problem I am having is how to downgrade the python version.  I'm not great with Linux and it is a pretty complicated process involving not only building the older version, but then setting it up as the default version.  I haven't found an easy way to either go back to an earlier python version or to even install and older Raspbian OS with the older python.  I did have an older image on hand and starting with it, upgraded it to the latest OS so that it would work on the pi zero 2 W.
If you have an older image of Raspian OS with the older python on it (and have a pi that it will work on), you can upgrade the OS and leave the older python:
sudo apt update && sudo apt upgrade -y
sudo apt clean

HFabi, 2021-11-18:
Maybe this helps you. I am currently testing it: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1298

TerryS719, 2021-11-19:
Thanks for pointing me to that issue.  My search didn't turn that up.  I ended up editing the cpp files:
rename tp_print to tp_vectorcall_offset
		in
		/bindings/python/rgbmatrix/core.cpp
		/bindings/python/rgbmatrix/graphics.cpp
as was suggested in that issue.  That worked well and was fairly easy to do.

The only problem I have now is that the image dims slightly every few seconds when I use a pi zero 2 W.  It works perfectly with the same build on a pi 4b.  I've tried all of the suggestions I can find for image dimming or flickering fixes, but none seem to affect the dimming I see.  It may be the pi zero 2 w just isn't up to it.  I thought maybe it would be OK since reports say it compares to the pi 3b.  If I run out of things to try I'll open another issue on this in case someone has some thoughts.

TerryS719, 2021-11-21:
Glad to report that the slight dimming problem that I was experiencing (about every 3 seconds) on a 64 x 64 matrix using a pi zero 2 W with the Raspberry pi OS builds (regular and lite) is NOT present with dietpi.

Raspberry pi OS: 32 bit released 2021-10-30
Raspberry pi OS Lite: 32 bit released 2021-10-30

DietPi_RPi-ARMv7-Bullseye
DietPi v7.8.2 : 00:36 - Sun 11/21/21

closed 2021-11-21 by TerryS719

sityware, 2021-12-01:
> 

it works fine on zero.
test --led-slowdown-gpio=2

Ensure PWM mode is enabled.
Ensure onboard audio is disabled.  dtparam=audio=off in /boot/config.txt
run the display as superuser
ensure this flag is NOT set --led-no-hardware-pulse

sityware, 2021-12-01:
> Thanks for pointing me to that issue. My search didn't turn that up. I ended up editing the cpp files: rename tp_print to tp_vectorcall_offset in /bindings/python/rgbmatrix/core.cpp [...]

it works fine on zero.
test --led-slowdown-gpio=2

Ensure PWM mode is enabled.
Ensure onboard audio is disabled. dtparam=audio=off in /boot/config.txt
run the display as superuser
ensure this flag is NOT set --led-no-hardware-pulse

davemaster, 2021-12-01:
Greetings,

When those problems exists, first, uninstall/remove any previous python version, then try to run sudo apt-get update, or search the right python version for You, sudo apt-cache search python, then sudo apt-get install Your_python_version

best regards

commit 45d3ab5d (2021-12-03): Genarate Python binding with latest cython.

Fixes #1367