hzeller/rpi-rgb-led-matrix #1371: No image with adafruit bonnet

Leon-NL, 2021-12-02:
I have checked the wiring and the software but I can't get the pi zero W 2 to work with it.  I am running the latest version of rasbian lite (bullseye). but when trying to start a demo my matrix stays black. I have a matrix (32x64) with the RUC7258. 
So if anyone has had the same problem and has/knows a solution please let me know.

hzeller, 2021-12-02:
Did you set the [right hardware mapping for the Adafruit Bonnet](https://github.com/hzeller/rpi-rgb-led-matrix#switch-the-pinout) ?

(also note that the Pi Zero W2 is also fairly new, so you need to have the rpi-rgb-led-matrix library version later than November 16th if you haven't updated yet)

Leon-NL, 2021-12-02:
Yes, I did add the hardware mapping to the settings when starting a demo.  I also was using the adafruit install script. But i did a fresh install of the software with git clone.  than i did "make -C examples-api-use" and after that "sudo examples-api-use/demo -D0 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=64 --led-chain=1" but still nothing.

hzeller, 2021-12-02:
I don't know what the Adafruit install script does, but if you have made sure to get the latest rpi-rgb-led-matrix from _this_ repository (careful, don't use the Adafruit repository as they usually have a somewhat older fork), then you should have a version that should work with the Pi W2 (I don't have one, but it was confirmed by owners of that Pi).

If this is something to do with your panel, it might be useful to test through the available `--led-row-addr-type` and  the `--led-panel-type` options.

Leon-NL, 2021-12-02:
I tried the settings for `--led-row-addr-type` and the `--led-panel-type` options but no succes. I did found out that my flat cable has a broken R1. I do not know if this can be a problem for the whole display. 
I have tried to use the display with a esp32 and the libary https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA and it worked right away.
also tried a pi zero W and same results as the pi zero 2

hzeller, 2021-12-02:
If RUC7258 is a chip that does things differently than the supported led panels, you might need to provide a patch that adds that support.
Where did you get the LED matrix from ? Do you have a link to the datasheet of that chip ?

Leon-NL, 2021-12-02:
link datasheet http://www.ruichips.com/uploads/file/20180816/20180816084036429.pdf but is almost entirely Chinese. it also has the TC5020A on the PCB as the LED drivers. I got the panel from a local electronics store in my area.

hzeller, 2021-12-02:
This does look fairly standard and it looks like it should work, but I don't have a Matrix of this kind (and no Raspberry Pi 2W) for this to test.
Do you have another Raspberry Pi lying around so that we can double-check that it is not due to the Pi ?

malani-art, 2021-12-03:
Hey, I'm facing the same issue. Although, I followed the install instructions from this github: https://github.com/DIYCharles/RGBGIFSlideshow

In any case, I'm using a Pi 3 A+ with Adafruit's 64x64 LED matrix and HAT. 

In the terminal, I run:
`~/rpi-rgb-led-matrix/utils $ sudo /home/pi/gifs/gifscript`

And it returns:
```
1
{Welcome 1}
Size: 64x64. Hardware gpio mapping: adafruit-hat-pwm
Loading 1 files...
Loading took 0.288s; now: Display.
```

But without any actual display... just a black matrix.

I know the matrix works and is fine though because when I run any of the ./demo, they show up perfectly.

Leon-NL, 2021-12-03:
@hzeller I just tried it with a 4B and I have the same results as yesterday. i used the same SD card with the command `sudo ./demo -D0 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=2`. 
When I measure the C and A pin i get a voltage of 1,6 volts.
I also turned of 1wire and i2c manually trough raspi-config

hzeller, 2021-12-03:
@malani-art  Please don't hijack another issue that has nothing to do with your problem.

You're using adafruit-hat-pwm: I suppose you did the necessary [hardware modification](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) for that ?

Also why do you use adafruit-hat-pwm and then say no hardware pulse ? Don't use `--led-no-hardware-pulse`, it makes the image quality bad.

hzeller, 2021-12-03:
@Leon-NL if you have powered the LED panel properly, the cable is intact and your adafruit board is working and you tried all the led panel types with the latest matrix library, then it could only be that your panel is not compatible. Next step is to pull out the oscilloscope and figure out what is going on. You mentioned you used another library that works and explicitly mentiones RUC7258 - maybe check the code there to see if they do something special for that ?

malani-art, 2021-12-03:
@hzeller Oops! Didn't mean to hijack - was intending to give insight as I'm facing the same issue and you asked the OP to test on another Pi.

Leon-NL, 2021-12-03:
@hzeller Thank you for your support. it was really quick and helpful. I will try to make something work and if I eventually find out how to get it to work I will put this in the thread.

Leon-NL, 2022-03-27:
After not trying for a while it just works now. i am using a pi4 with the latest version of bullseye (64bit). so i dont know what the problem was.

Leon-NL, 2022-05-13:
tried it on the pi zero 2W and it works. so i will close this thread

closed 2022-05-13 by Leon-NL