hzeller/rpi-rgb-led-matrix #1373: Black Matrix - nothing shows up

malani-art, 2021-12-03:
Hey, 

Making my own thread here. I've been having this issue all day. I tried different githubs and they all return the same output (none). So I decided to delete all of those, re-install this git, and start over. Even with this git, I am faced with a black screen.

I am using a Pi 3 A+ with Adafruit 64x64 matrix and HAT. Demos do light up the matrix so I know there is no issue with my setup. I wonder if it is an issue with my sudo command or perhaps my .gif?

In any case, my matrix does not light up even though the terminal says it is displaying when I run 
`pi@Malani:~/Desktop/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer -C /home/pi/gifs/1.gif --led-no-hardware-pulse --led-cols=64 --led-rows=64
`
Note: I've included `--led-no-hardware-pulse` because without it, i get an error about sound module (although I turned it off in config.txt... so I'm not sure what is going on there)

In any case, I am returned with:
```
Suggestion: to slightly improve display update, add
	isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
Size: 64x64. Hardware gpio mapping: regular
Loading 1 files...
Loading took 0.290s; now: Display.
```
But nothing displays. At all. :(

I'd really appreciate some help!

davemaster, 2021-12-03:
Greetings,

What about your wiring/connections, can You share yours?

hzeller, 2021-12-03:
If you use the adafruit hat, make sure to choose relevant `--led-gpio-mapping` as described in the README.
The output shows that you use the regular hardware mapping, not the adafruit one.

malani-art, 2021-12-03:
> If you use the adafruit hat, make sure to choose relevant `--led-gpio-mapping` as described in the README. The output shows that you use the regular hardware mapping, not the adafruit one.

Running the sudo with --led-gpio-mapping=adafruit-hat made it work! Oh my goodness! Thank you! I should've known it was a simple fix...

Although, strange that although I turned off sound module via config.txt, I still receive warning about it when sudo. So, I'm forced to also add --led-no-hardware-pulse. This works for now but I wonder if you have any fixes for that. 

In any case, thank you :)
--
Log:

```
pi@Malani:~/Desktop/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer -C /home/pi/gifs/1.gif --led-gpio-mapping=adafruit-hat --led-no-hardware-pulse --led-cols=64 --led-rows=64
Suggestion: to slightly improve display update, add
	isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
Size: 64x64. Hardware gpio mapping: adafruit-hat
Loading 1 files...
Loading took 0.293s; now: Display.

```

hzeller, 2021-12-03:
Did you reboot the Pi after the config change ? Usually the config change should work (and also make sure to do the isolcpus setting as described in the readme).

If for whatever reason the sound config doesn't work with the config file, maybe just blacklist the sound module (if you search the internet how to disable sounds on the Raspberry Pi, you find some suggestions)

closed 2021-12-03 by hzeller