hzeller/rpi-rgb-led-matrix #1375: 64x64 Panel + Adafruit Bonnet + Raspberry Pi Zero WH: Blank Rows/Columns

tward212, 2021-12-06:
Hey all,

I've been troubleshooting an issue for a while with getting the demos to run on a 64x64 LED Matrix from Adafruit using a Adafruit Bonnet on a pi zero WH running Raspbian Lite.

I went back and forth with Adafruit support for a while and wasn't able to solve the issue (see the following forum discussion: https://forums.adafruit.com/viewtopic.php?f=47&t=184987&p=900296#p900296)

TL;DR version is that I've gone through about three Bonnets and triple checked all the connections/hardware in my setup.  I've soldered the E-line and a jumper to GPIO 4 and 18.  I've tried multiple variations on the parameters including --led-gpio-slowdown=1, --led-multiplexing=1, --led-gpio-mapping=adafruit-hat, --led-row-addr-type, etc. but there is no significant change.

I'm thinking there is something on the software side that I'm missing, but I'm completely stumped on what my next steps should be.

This is my first project and I'm using it as a way to learn basic soldering, code, etc. so sorry if I've failed to explain something.  I'm very much a beginner when it comes to this type of thing, so any help would be greatly appreciated.

Thanks!

(also, I've tried the solutions suggested in #895 and other linked issues, but nothing that has worked so far)

Below is an example image 
Note: this was taken while the demo was rotating so ignore the black spaces where the leds are on.  That's just because of the camera capture.  The real blank spot is that gap in the middle. Similar in appearance to the matrix in #1279 

[image]

sforment, 2021-12-14:
I had this same issue with my 64x64 matirx and the Electrodragon board and was pounding my head against the wall. I tried everything, but eventually what worked wasn't adding my parameters but adding less. Try cutting parameters bit by bit leavign only the most necessary parameters. In your case I would assume that the Adafruit hat specific parameters will be needed, but don't add anything else.

sforment, 2021-12-14:
Try using just these flags:

--led-cols=64 --led-rows=64 --led-gpio-mapping=adafruit-hat

sforment, 2021-12-14:
"If you have a Raspberry Pi with a slower processor (Model A, A+, B+, Zero), then a value of 0 (zero) might work and is desirable.":

This is also noted in the LETS DO THIS instructions. so you may need the following parameter, but I would try it first without this to see if the default settings work:

--led-slowdown-gpio=0

tward212, 2021-12-14:
@sforment Thanks for the reply.  Glad to know it's not just this particular set of hardware that can have this issue.

I have messed around with different combos of parameters, but I'll give that a shot again.   It's interesting because I tried this on multiple sets of hardware and the issue persists, so it's definitely something in the software or code that is causing this.

My next step is going to be starting from square one with a new OS install, etc. and then more thoroughly document what I try/what does or what doesn't work. 

Anyway, I'll drop some more info in here once I'm able to get some time to take a second look at everything.

sforment, 2021-12-14:
I would also highly recommend using the Electrodragon breakout board. I've ordered twice and shipping takes quite a while, but ultimately worth it since the Adafruit board seems harder to work with from what I gather reading through issues posted here.

Maybe try that route before giving up.

I'm using DietPi as my OS also. Worth a shot trying that as well I guess if you run into more problems. 

Another things I've read here is that the matrixes are unpredictable. They seem to change build/configuration types without notice, but seeing how you ordered this from Adafruit I think that is not the issue. Let me know if you have any better results with these options. I'm no expert, but I got mine working through trial and error and I'm sure the solution is out there with the hardware you have.

tward212, 2021-12-16:
@sforment I'm working with DietPi right now and I'm hitting some weird snags.  Specifically, once I get into the examples-api-use directory and try to execute any `sudo ./demo ....` code I get hit with the error: "sudo: ./demo: command not found"  

I'm sure there is something really simple I'm missing, but I'm struggling to figure it out.

hzeller, 2021-12-16:
Did you type `make` ?

tward212, 2021-12-17:
@hzeller thanks, it was that and I also needed to run `sudo apt-get install build-essential` 

I was also typing `make` in /examples-api-use, but I guess I needed to type it in /rpi-rgb-led-matrix?  Either way it should be good now.

Also, just waited a bit and let everything install.  Hit `sudo ./demo -D0 --led-cols=64 --led-rows=64 --led-gpio-mapping=adafruit-hat` and the demo is running perfectly.  Looks like switching to DietPi was the solution.

Thanks @sforment and @hzeller !

closed 2021-12-17 by tward212

sforment, 2021-12-17:
Glad it worked out for you @tward212 ! Cheers!