hzeller/rpi-rgb-led-matrix #1378: 2 pixels light on, instead of 1

olmichal, 2021-12-15:
Hello everyone!

I have a problem with the pixel display. When I use the SetPixel() function, the corresponding pixel is displayed and one on the right. For example: I set the SetPixel function `SetPixel(1,1,255,255,255)` and the matrix displays pixel `[1] [1]` and `[2] [1]` in white. Or in the rectangle example, there are 11 columns instead of 10.

Here is my code: [test1.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/7721166/test1.txt)

I cloned the repo:
`git clone https://github.com/hzeller/rpi-rgb-led-matrix`

I have also installed some dependencies:
`sudo apt-get update  && sudo apt-get install -y git python3-dev python3-pillow`

I used those commands in the repo folder:
`make build-python PYTHON=$(which python3)`
`sudo make install-python PYTHON=$(which python3)`

I have made my code executable:
`chmod +x test1.py`
`sudo ./test1.py`

Here is how I have made the wiring:
| Connection | Pin | Pin | Connection |
| ---: | :---: | :---: | :--- |
| - | 1 | 2 | - |
| - | 3 | 4 | - |
| - | 5 | 6 | GND |
| STB | 7 | 8 | - |
| - | 9 | 10 | - |
| CLK | 11 | 12 | OE |
| G1 | 13 | 14 | - |
| A | 15 | 16 | B |
| - | 17 | 18 | C |
| B2 | 19 | 20 | - |
| G2 | 21 | 22 | NC |
| R1 | 23 | 24 | R2 |
| - | 25 | 26 | B1 |

I also set `isolcpus=3` in cmdline.txt and `dtparam=audio=off` in config.txt

**Is there anything I can do to fix this?**

[image]

[image]

MichalLinde, 2022-01-23:
I have the exactly same problem

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828