hzeller/rpi-rgb-led-matrix #1379: 128x32 Matrix

arconkli, 2021-12-18:
Hey, I'm trying to get my 128x32 matrix setup and when I use the flags: **--led-parallel=1 --led-chain=2 --led-cols=64 --led-rows=32 --led-show-refresh --led-limit-refresh=0 --led-pwm-lsb-nanoseconds=175** nothing shows up, but when I change the "--led-parallel=1" to "--led-parallel=2" and "--led-rows=32" to "--led-rows=16" I get something that looks like this
[image]
So I am just unsure why the original flags aren't working and just looking for some help with that, thank you.

arconkli, 2021-12-18:
It seems to only show an output when the led-parallel is 1>.

arconkli, 2021-12-18:
This is all I've got to somewhat work with the "runtext" however it doesn't work with the rotating square as it puts it to the very left side in the "middle" of the 256x32 **sudo ./demo -D1 --led-parallel=2 --led-chain=1 --led-cols=128 --led-pixel-mapper=V-mapper --led-show-refresh --led-limit-refresh=0 --led-no-hardware-pulse --led-row-addr-type=0 --led-slowdown-gpio=3  runtext16.ppm**

davemaster, 2021-12-18:
Greetings
If You don't know what type of RGB PANEL have, try with different values of --led-multiplexing=XXX   

good luck