hzeller/rpi-rgb-led-matrix #1381: 128x128 ( 2 x 128x64? ) panels with LS9929 and SM5216 shift-registers

hedj, 2022-01-01:
G'day. 
  I obtained a couple of the panels below from china with the thought in mind of building an alarm clock for my son. They have LS9929 and SM5216 shift registers for columns and rows.
The silkscreen says they're a 1/32-scan panel and the connectors say HUB75E. When asked for pinouts, the vendor gave me
a picture of a HUB75 pinout; I'm not exactly sure which is correct.

I'm using an Adafruit hat for a rpi4, but have thus far no success in turning on even a single pixel. I've tried different multiplexing and row address setting parameters to no avail. Also tried again manually setting pin values using an arduino. I have verified power to the device and that I can indeed toggle signals between 0 and 5V at the connector. I do have access to an oscilloscope; is there a suggested process for debugging the panel driving? I have no datasheet for the LS9929 chips, though the 5216 is here:

http://www.88869188.com/uploads/201801/5a542100d93e6.pdf

Any help greatly appreciated.

[image]

hedj, 2022-01-01:
It looks like the LS9929 is probably the cause of the troubles; the manufacturer blurb mentions SRAM, which is ominous.
Looks like serial data in for row select also. Anyone know how to initialise these chips?

[image]
[image]

hallard, 2022-01-02:
Not sure how to help but would be interested by these panels, where did you get them?

hedj, 2022-01-03:
Bought from here:

https://ledlightsworld.com/collections/small-pixel-pitch-led-modules-p1-25-p2

Sent from ProtonMail for iOS

hallard, 2022-01-04:
thanks, looks like they are not available anymore. Anyway price is expensive, I succeeded to assemble 2 128x64 panels, cheaper and does works fine :-)

daniel-live-update, 2024-01-14:
Hi hedj, I know this goes back 2 years but I was wondering if you ever can right?  I've built a mini board to control HUB displays and I can get almost all to work except ones with the LS99xx chips.  I also can't find a datasheet other that the block diagram you also found.

MuttonBird, 2024-12-13:
Hi @daniel-live-update , so were you able to get any info on LS99xx chips? I got one board with that chip and I couldn't get to turn it on either, thanks.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828