hzeller/rpi-rgb-led-matrix #1394: My code isn't working!

SnowyDevelop, 2022-01-22:
I have gotten a code from jawedsoft (https://github.com/hzeller/rpi-rgb-led-matrix/issues/1028) but it my matrix is black and i dont see anything!

This is the command i'm using`sudo python jonasuhr.py --led-chain=4 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat`

And Im getting this:
[image]
How can i fix it?

hzeller, 2022-01-22:
Start with the simple things: do the c++ demos work (with adafruit-hat mapping), then do the included Python demos work; then compare your code to where things differ. Do rubber duck debugging.

In general, it is hard to remote diagnose code that is on someone elses computer :) You wrote it, you will be able to fix it. If you have not written it (sounds like you used part of jawedsoft), make sure to first understand all the parts of that software.

SnowyDevelop, 2022-01-22:
Ok thanks!