hzeller/rpi-rgb-led-matrix #1396: When Running Command text-example show unreadable Text

b4m84, 2022-01-26:
Dear @hzeller , i have question when running this command :
#sudo ./text-example -f ../fonts/5x8.bdf  --led-no-hardware-pulse

in P10 display show unreadable Text like =
[image]

and this is my wiring configuration =
[image]

Maybe any solution for this ? i have change font , it's same show unreadable text.

Note: When i run demo =
#sudo ./demo -D0 --led-no-hardware-pulse  
it's running and show like this =
[image]

Thanks
Bambang

hzeller, 2022-01-26:
Looks like you need to choose a [`--led-multiplexing`](https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays) that fits your display.

b4m84, 2022-01-26:
Thanks @hzeller it's work when i set --led-multiplexing=8
sudo ./text-example -f ../fonts/6x13.bdf  --led-no-hardware-pulse --led-multiplexing=8
[image]

thanks for your support

closed 2022-01-26 by b4m84