hzeller/rpi-rgb-led-matrix #1397: Cannot load fonts

genregasm, 2022-01-26:
`pi@piLED:/home $ sudo python3 pi/rpi-rgb-led-matrix/bindings/python/samples/runtext.py --led-cols=128 --led-gpio-mapping
=adafruit-hat --text="Hello, world!"
Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
Press CTRL-C to stop sample
Traceback (most recent call last):
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/runtext.py", line 35, in <module>
    if (not run_text.process()):
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 76, in process
    self.run()
  File "/home/pi/rpi-rgb-led-matrix/bindings/python/samples/runtext.py", line 16, in run
    font.LoadFont("../../../fonts/7x13.bdf")
  File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
Exception: Couldn't load font ../../../fonts/7x13.bdf`

hzeller, 2022-01-26:
since you're running that from a different directory than the samples directoy, you've to adapt the path to the font (just edit the file)
Probably just make them absolute paths to where the fonts are.

genregasm, 2022-01-26:
New to python, but learning, thank you.

closed 2022-01-26 by genregasm