hzeller/rpi-rgb-led-matrix #1398: There is no signal at outputs P2 and P3, the Electrodragon bonnet

Taki75, 2022-01-26:
panel type: P2.5 FM6127 64 * 64
bonnet: https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

First of all, I would also like to thank you for all the work and the much help you have given me. I never thought I would get this far, but it looks like a lot of learning and experimentation has paid off.
My first problem was that there were blank lines left on the 64 * 64 display (16y16n16y16n) as here in the video:

https://www.youtube.com/watch?v=Y95ItZfQivA

[image]

But by switching this switch, the problem was resolved:
[image]

In fact, everything went well and goes well since then (demo D4, D7, text-scroller, etc.) if I drive a 6 * 64 * 64 panel on the P1 connector, but if I want to play video with the video-viewer , it's all a big flicker, whatever setting I'm running. I guess the problem here may be that I drive more than 24k LEDs on a single strand. I read that on a thread approx. The limit is 10k if you want to achieve a normal resolution and refresh rate.

In case I divide this by putting three panels on P1 and three panels on P2, only P1 will display image, not P2. The same is true if I put 2-2 panels on 3 channels, only P1 works P2 and P3 does not. If I put the line on P2, all I get is a blue light on the 64 * 64 display at 64 * 32 pixels.
[image]

I run the following command when I get relatively normal operation:

https://www.youtube.com/watch?v=VsPd8Uk9q-M

https://www.youtube.com/watch?v=CPcAjbM3_WE

sudo ./demo -D10 --led-rows = 64 --led-cols = 64 --led-chain = 6 --led-parallel = 1 --led-panel-type = FM6127 --led-slowdown-gpio = 5
I am still modifying these values, but the same error:
--led-chain = 6 --led-parallel = 1
--led-chain = 3 --led-parallel = 2
--led-chain = 2 --led-parallel = 3

I also ran this to see more examples:

sudo ./text-scroller -f ../fonts/texgyre-27.bdf -B0,0,255 -O0,0,100 -C255,0,0 --led-chain = 4 "Hello ♥" --led-rows = 64 --led-cols = 64 --led-chain = 6 --led-panel-type = FM6127 --led-slowdown-gpio = 5 --led-brightness = 20 -y10

I was sent a bit of a strange bonnet by Electrodragon, as I saw in several pictures that there are 6 jumper options to connect the lines, but I have 8 needles. Maybe I don't tie them in well? I also ordered two bonnets, both have the same problem.

I tried these connections, but none of them are good:
[image]
[image]

Thank you for your concern.

xuniuer, 2022-04-01:
I use another version board (https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/), just to connect P4 with e_buf, it works for 64x64 panel, QiangLong P3 with icn2013ep chip embedded.

faizalsrahman, 2022-05-16:
Hello @xuniuer, can you please help me in setting up raspberry pi4 with the board that you have used(https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/). I bought the same one and connected to 32x32 led panels chained into 4 and paralleled to 3

xuniuer, 2022-05-20:
@faizalsrahman More information needed to provided to check possible problems.

faizalsrahman, 2022-05-30:
@xuniuer  Please find the below additional details,
I followed the README files of [GitHub - hzeller/rpi-rgb-led-matrix: Controlling up to three chains of 64x64, 32x32, 16x32 or similar RGB LED displays using Raspberry Pi GPIO 1](https://github.com/hzeller/rpi-rgb-led-matrix) and wired the things, I can able to see the unclear text/image while trying to run following commands " sudo python image-scroller.py --led-cols=64 --led-rows=32 --led-chain=2". Kindly help me to identify the issue and fix it.

[https://youtu.be/fVvg565Gh1Q 5](https://youtu.be/fVvg565Gh1Q)

I also get the reference from [https://iot-for-maker.blogspot.com/2020/01/led-5-lets-make-large-led-display-part_84.html 1](https://iot-for-maker.blogspot.com/2020/01/led-5-lets-make-large-led-display-part_84.html)

Hardware Details:
Raspberry Pi 4
[Electrodragon HAT V2](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-for-raspberry-pi-v2/)
[LED Panel](https://www.indiamart.com/proddetail/qiangli-p6-outdoor-led-module-21348696691.html)

xuniuer, 2022-05-31:
Sorry that I could not view the references above for local Internet regulations policy. 

However, here are some suggestions.
1. To use `demo`  firstly to check necessary parameters or configurations, then use python to go ahead;
2. `multiplexing` is the most important parameter to config panels, especially, when your panels could not run out of box;
3. If you need a more screen, to start with one panel, make it work, then extend to chains or parallels...