hzeller/rpi-rgb-led-matrix #1400: MPV player conflicting with LED matrix

creating-worlds, 2022-01-27:
Hi there,

first of all thank you for this amazing library. It works like a charm!

My setup: 
```
    options.rows = 32
    options.cols = 704
    options.chain_length = 1
    options.hardware_mapping = 'adafruit-hat-pwm'
    options.gpio_slowdown = 3
    options.brightness = 100
    options.pwm_lsb_nanoseconds=100
    options.pwm_bits = 7
    options.pwm_dither_bits = 0
```

Hardware:

```
Raspberry Pi 4
Raspberry Pi OS Bullseye (Headless)
Adafruit Bonnet with PWM fix
```

I am trying to play a video on an additional screen via HDMI (**I am NOT trying to output a video on the Matrix**). I am using the MPV Player ([python-mpv](https://github.com/jaseg/python-mpv)) to output the video.

Here is my code (at least the important bits)

```
import mpv
import time
import threading
import socketio
import json
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics

class SIO_Wrapper():
    current_channel = 0
    sio = socketio.Client()
   .......
    [MY SOCKETIO FUNCTIONS]
   .......

class VideoPlayer():

    def start_video_in_thread(self, video_name):
        thread = threading.Thread(target=self.play_video(video_name))
        thread.start()

    def play_video(self, video_name):
        print("start video")
        player = mpv.MPV(vo="gpu", gpu_context="drm", hwdec="rkmpp")  
        player.play("/path/to/vid/{}.mp4".format(video_name))
        player.wait_for_playback()
        print("stop video")

class MatrixManager():
    
    options = RGBMatrixOptions()
    options.rows = 32
    options.cols = 64
    options.chain_length = 1
    options.hardware_mapping = 'adafruit-hat-pwm'
    options.gpio_slowdown = 3
    options.brightness = 100
    options.pwm_lsb_nanoseconds=100
    options.pwm_bits = 7
    options.pwm_dither_bits = 0

    def __init__(self, *args, **kwargs):
        super(MatrixManager, self).__init__(*args, **kwargs)

    def initialize(self):      
        matrix = RGBMatrix(options = self.options)
        return matrix

if __name__ == "__main__":
    video_player=VideoPlayer()
    socket = SIO_Wrapper()
    socket.run()
    led_matrix = MatrixManager()
    matrix = led_matrix.initialize()

    while True:
        channel = socket.get_current_channel()
        if channel == 1:
            video_player.start_video_in_thread("TESTVIDEO")
            socket.set_current_channel(0)
            print("channel 0")   
```

First I thought this might be a threading issue with my socketio client but it turns out that there is a conflict with the matrix library. If I comment out **matrix = led_matrix.initialize()** the video plays perfectly. However, when the matrix was initialized before I try to play the video, it prints "start video" and without any output it prints "stop video" immediately after.

Do you have any idea why this is happening?

hzeller, 2022-01-28:
I suspect you probably run into permission issues: when the matrix starts, it changes the permissions from `root` to an unprivileged user to minimize attack surface. However, if you then open files (such as, say, a movie file), that file needs to be able to be read by the new user (which is `daemon`).

So either make the file `chmod a+r videofile.mp4` or set the runtime options of the matrix to not `drop_privileges`.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828