hzeller/rpi-rgb-led-matrix #1406: P2.5 320x160 hub75E not turning on

ignaciogaticarojas, 2022-02-14:
Hello @hzeller . Thank you so much for this amazing library. Ive been enjoying it more than a year now...

I just got this displays from aliexpress[ ](https://es.aliexpress.com/item/1005002038749262.html?gatewayAdapt=glo2esp&spm=a2g0o.9042311.0.0.7fb163c0XIjDxd) and i cant make them turn on. I have a 5v amp. a rasberry pi bonnet a raspberry pi 3 . 

Could you please let me know if this is due to the E line or is it something else? Is my adafruit Bonnet the problem?

Thanks a lot

ignaciogaticarojas, 2022-02-15:
this are the chips and specs that the seller sent
[P2.5 flexible module Specification (2).pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/8065028/P2.5.flexible.module.Specification.2.pdf)
[image]
[image]

ignaciogaticarojas, 2022-02-17:
HEEEELP

hzeller, 2022-02-17:
I don't have any panel with that chip, so I can't test it and thus implement such a feature.

ignaciogaticarojas, 2022-02-17:
Omg thanks for your quick reply! After using your program im totallya fan of your work.

I could send u one if you like to test it one day . 
For now im going to use classic p5 ones . Building a nice stockmarket ticker display program for the exhibition. Is going to be held in Hessel Museum of Bard University.
Again thanks for all your amazing work.

All the best

Ignacio Gatica

closed 2025-11-27 by marcmerlin