hzeller/rpi-rgb-led-matrix #1408: Long single chain of 22 panels 32x64 glitch on last 2 panels on raspberry pi 4

ignaciogaticarojas, 2022-02-25:
Hello again @hzeller !
I decided to continue my project with other panels than the ones i bought in china.

Im running a the program on 22 32x64 p5 panels with a raspberry 4 with adafruit bonnet with gpio:4 and it still glitching on the last two panels.

is there any recommendation on how to fix this?

(i also have a electrodragon hat but never learned how to make it work in one single chain of if this one would make any difference)

https://user-images.githubusercontent.com/99695299/155749121-e628939a-47e2-4a3c-805c-a3055f1150ca.mp4

davemaster, 2022-02-25:
22 panels chained are too much. Try to splity that chain in half, 11-11 ie, two parallel or, three parallel of 7 each (one of 8 panels)

ignaciogaticarojas, 2022-02-25:
Thanks @davemaster , can i ask you what are the parameters an changes i would have to do in order to achieve this with electrodragon instead of raspberry hat? also i do need a very long  data cable for that right?

thanks a lot

davemaster, 2022-02-25:
[image: image.png]

You have to build a large cable for the second half

add :

--led-parallel = 2; --led-chain=11 (if this is the half of panels)

The problem, is find out the way to use the mapping using the option of the
rpi-rgb-led-matrix option, You check it out.

Best regards.

ignaciogaticarojas, 2022-02-25:
Thanks a lot @davemaster , Dear @hzeller is it possible you could suggest me what is the best option for mapping this long ticker chain? 

Again thanks a lot

hzeller, 2022-02-25:
If there is only a slight skew between the passed through clk signal and the output of the shift register, then you'll get artifacts such as that.

Parallel chains with 3x8 or 2x11 is recommended (`--led-chain=11 --led-parallel=2` or `--led-chain=8 --led-parallel=3` - in the latter case, you are not using two of the addressed panels, so maybe extending your number of panels to 24 might be cool). To arrange them all phisically in one line, use the [V-mapper](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#v-mapper-and-vmapperz-vertical-arrangement)

In general, this will also increase your refresh rate.

hzeller, 2022-02-25:
also gpio slowdown is probably a good first defense, but you guess you read the readme and have done that already.

ignaciogaticarojas, 2022-02-26:
thank you for the quick reply @hzeller ( yes to gpio readme question) ---- now i step into a probably baby step problem, but following this instructions:  [i](https://iot-for-maker.blogspot.com/2020/02/led-3-lets-make-large-led-display-part.html) the electrodragon on my raspberry 4 plugged in the first left hub75 of the hat show me this

https://user-images.githubusercontent.com/99695299/155819134-fb11a407-8802-4473-87d9-2b32d124c2f7.mp4

this is the command im using sudo

 python3 image-scroller.py  --led-cols=64 --led-rows=32 --led-gpio-mapping=regular

hzeller, 2022-02-26:
you might also need `--led-gpio-slowdown=4` or so - the Pi4 might just be too fast for your panels.
A single chain on the electrodragon should look like a single chain elsewhere with the difference of the led-gpio-mapping in parameter.

ignaciogaticarojas, 2022-02-26:
thanks @hzeller feeling like im almost there , but now its looking like this

https://user-images.githubusercontent.com/99695299/155820117-68a13254-5359-400f-82d4-41c7e3393d97.mp4

at least i see something ahaha !

ignaciogaticarojas, 2022-02-26:
is this a pin i have to solder in the electrodragon hat? @hzeller 
thanks again

hzeller, 2022-02-26:
I don't know the electrodragon hat, you should follow their instructions

ignaciogaticarojas, 2022-02-26:
@davemaster are you familiar with this kind of black horizontal lines?

ignaciogaticarojas, 2022-02-26:
is this usually related to some gpio thing?

maksun, 2022-02-26:
Hello,

The electrodragon hat the same as the "hzeller" hat, I think they did a hat with the instructions we can find here.
That's interesting as someone reported me kind of the same issue:

[image]

He also told me it works with another hat (joy-it) but from what I saw it is also a clone of "hzeller" hat.
I'm waiting for more information to understand what is going on, I will inform you here if I have something interesting.

Regards,

ignaciogaticarojas, 2022-02-26:
@maksun i found this in the electrohat website 

Matthew
The problem is on the board. See channel D. It’s grounded. I have seen this problem on my boards. Cut the track and solder a wire between pin 12 on panel 0 to pin 12 on panel 1.

ignaciogaticarojas, 2022-02-26:
[image]
[image]

i hope this help some electrodragon users... now im going to proceed with the surgery

ignaciogaticarojas, 2022-02-26:
if there is any suggestion on how to cut this would be highly appreciated @hzeller @davemaster @maksun

davemaster, 2022-02-26:
Look, when I still not have the electrodragon HAT for the raspberry, I used JUST CABLEs, those female-male cables to wiring to the rgb panel INPUT. Do as I did, use just cables, for TWO chains; as THE "regular" connection.

Details: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

Best regards

maksun, 2022-02-26:
@ignaciogaticarojas Great you find this info, thank you :) You can take any sharp and thin blade (like the one for surgery or a
little cutter with a new blade). Please tell us if it is OK after the "surgery" :)

maksun, 2022-02-26:
@ignaciogaticarojas @davemaster @hzeller Just for your information - if someone is asking how to fix - it looks like removing
only the two "holes" is enough then put a little bit of nail polish to avoid future corrosion:

Before:
[image]

After (the nail polish has not been put on this picture):
[image]

Then it works ;)

ignaciogaticarojas, 2022-02-26:
thats  cool @maksun i ended up building the aciva adapter that @hzeller designed . but i have a new question, it seems that the second and 3 hub dont light up . i already tried the following 
udo python3 image-scroller.py  --led-cols=64 --led-rows=32 --led-gpio-mapping=regular --led-slowdown-gpio=5 --led-chain=4 ----led-chain=3 ( also tried --led-chain=3)

any lead on how to make them work would help , thanks

davemaster, 2022-02-26:
I had purchased this:

[https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/](url)

ignaciogaticarojas, 2022-02-27:
would i need to build a custom mapper for i want to try to achieve? 
@davemaster @hzeller

Taki75, 2022-02-28:
> Hello,
> 
> The electrodragon hat seems identical to the "hzeller" one, I think they did a hat with the instructions we can find here. That's interesting as someone reported me kind of the same issue:
> 
> [image]
> 
> He also told me it works with another hat (joy-it) but from what I saw it is also a clone of "hzeller" hat. I'm waiting for more information to understand what is going on, I will inform you here if I have something interesting.
> 
> Regards,

if this fault occurs on the electrodragon panel, there is a switch on the panel, it must be switched and then the problem will disappear.

[image]

[image]

davemaster, 2022-02-28:
The problem, the REAL problem is that the hobbysts doesn't read the manuals, try to do the things in few minutes, even those more complicated.

PLS, fist, read the repository README.md, in this, is full detailed (wiring included, schematics included)

maksun, 2022-02-28:
@Taki75 The switch is here to use port 3 OR the RTC clock, the problem we are talking about is not linked, even electrodragon
says: "Yes indeed there is a problem, The line D (pin12 ) is failed and grounded by a design mistake, please cut the line as in
the attached image, if you need a replacement please contact us".

ignaciogaticarojas, 2022-02-28:
Hi David i read it many times and been working on this program for more
than a year and still finding interesting issues ! anyways thei figured out
and what you suggested works. Thanks again everyone ![image: WhatsApp Image
2022-02-28 at 1.25.52 PM.jpeg]

Bano310, 2022-04-05:
> [image] Tienes que construir un cable grande para la segunda mitad agregar:
> […](#)
> --led-paralelo = 2; --led-chain=11 (si esta es la mitad de los paneles) El problema, es encontrar la forma de usar el mapeo usando la opción de la opción rpi-rgb-led-matrix, échale un vistazo. Atentamente. El vie, 25 feb 2022 a las 14:11, ignaciogaticarojas (< ***@***.***>) escribió:
> -- David Elias Flores Escalante TeleTracking SAC

[](https://github.com/xuniuer)[](https://github.com/Bano310)[](https://github.com/Bano310)[](https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)[](https://github.com/Bano310)[](https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)[](https://github.com/xuniuer)[](https://github.com/Bano310)[](https://github.com/)Mi código está parpadeando, ¿cómo puedo eliminar este parpadeo?

este es mi código
importar json
desde samplebase importar SampleBase
desde rgbmatrix importar gráficos
importar RPi.GPIO como GPIO
importar tiempo
importar sys

GPIO.setmode(GPIO.BCM)
GPIO.setup(21, GPIO.IN)
GPIO.setup(24, GPIO.IN)
GPIO.setup(18, GPIO.OUT, initial=0)

cont =100
plazas =100
clase PruebaGráfica(SampleBase):
def init (self, *args, **kwargs):
super(GraphicsTest, self). init (*args, **kwargs)
def run(self):
canvas = self.matrix
font = graphics.Font()
font.LoadFont("../../../fonts/texgyre-27.bdf")

    green = graphics.Color(0, 255, 0)
    graphics.DrawText(canvas, font, 83, 24, green,str(cont))

    blue = graphics.Color(0, 0, 255)
    graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
    time.sleep(1)
while cont <= 100:
inputValue = GPIO.input(21)
inputValue1 = GPIO.input(24)
if (inputValue == True):
cont = cont - 1
print("Choche entrada " + str(cont))
time.sleep (.01)
class GraphicsTest(SampleBase):
def init (self, *args, **kwargs):
super(GraphicsTest, self). init (*args, **kwargs)
def run(self):
canvas = self.matrix
font = graphics.Font()
font.LoadFont("../../../fonts/texgyre-27.bdf")

            green = graphics.Color(0, 255, 0)
            graphics.DrawText(canvas, font, 83, 24, green,str(cont))

            blue = graphics.Color(0, 0, 255)
            graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
            inputvalue = False
        time.sleep(3)       
time.sleep(1)

if (inputValue1 == True):
    cont = cont + 1
    print("Salida" + str(cont))
    time.sleep(.01)
    class GraphicsTest(SampleBase):
        def __init__(self, *args, **kwargs):
            super(GraphicsTest, self).__init__(*args, **kwargs)
        def run(self):
            canvas = self.matrix
            font = graphics.Font()
            font.LoadFont("../../../fonts/texgyre-27.bdf")
    
            green = graphics.Color(0, 255, 0)
            graphics.DrawText(canvas, font, 83, 24, green,str(cont))

            blue = graphics.Color(0, 0, 255)
            graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
            inputvalue1 = False
        time.sleep(1)       

time.sleep(1)   
if (cont == 0):
    GPIO.output(18, 1)
    print ("parking lleno" +  str(cont))
    class GraphicsTest(SampleBase):
        def __init__(self, *args, **kwargs):
            super(GraphicsTest, self).__init__(*args, **kwargs)
            self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel")
        def run(self):
            canvas = self.matrix
            font = graphics.Font()
            font.LoadFont("../../../fonts/texgyre-27.bdf")
            LedChain = 4
            
    
            red = graphics.Color(255, 0, 0)
            graphics.DrawText(canvas, font, 0.5, 23, red,"completo")
        time.sleep(.3)
time.sleep(2)

    
if __name__ == "__main__":
    graphics_test = GraphicsTest()
    if (not graphics_test.process()):
        graphics_test.print_help()
tiempo.dormir(5)

 WhatsApp.Video.2022-03-30.a.11.48.40.AM.mp4

maheeetaaa, 2022-06-13:
> sudo python3 image-scroller.py --led-cols=64 --led-rows=32 --led-gpio-mapping=regular --led-slowdown-gpio=4 --led-chain=3

THANK YOU!!

Monishamacharla, 2022-06-14:
I had the same issue before but now it is working fine, I connected 16 panels 32x32 with single chain, and u mapper, just try with gpio slow down = 5, which you can configure in the led matrix.cc file.

ignaciogaticarojas, 2022-08-09:
Thank you @hzeller @marcmerlin @davemaster
https://user-images.githubusercontent.com/99695299/183772462-0174f387-7c7f-47d8-8a68-ba1e6fa3a1d1.mp4

ignaciogaticarojas, 2023-01-04:
Hello and happy new year community  @hzeller .
I found a problem that im guessing it could be power one building a 14 screen display powered by an 5v 70 amp power supply. Half of the panel chain glitches after a while being powered, Im guessing it could be because of the thickness of the cables that power the pi?

It would be great to get some input on this.
[image]
[image]
[image]
[image]

Here i attach images of the set up

anandrajgupta, 2023-01-05:
FRC of the Panel from where the Glitch starts 🤔

ignaciogaticarojas, 2023-01-05:
No its not the data cable. We changed it and still has the same problem... :/

anandrajgupta, 2023-01-05:
Yes ideally it should, but that should not be the actual reason behind this glitch.

Can you diagrammatically show the arrangement of Panels you have established, and also the matrix configurations corresponding to it, specially the GPIO SLOWDOWN parameter?

ledvinap, 2023-01-05:
Can you test that only half connected by longer cable is glitching?

Try installing jumper on P7-P8 first - ground line near RGB data may help.

ignaciogaticarojas, 2023-01-05:
Try installing jumper on P7-P8 first - ground line near RGB data may help. where is this? @ledvinap in adaptor or panels?

@anandrajgupta here is a diagram of the set up

and configuration is app.py led-cols=448 --led-slowdown-gpio=5 --led-gpio-mapping=regular  --led-rows=32 --led-chain=1 --led-pixel-mapper=V-mapper  --led-parallel=2 --led-brightness=70

[setup la.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/10353801/setup.la.pdf)

anandrajgupta, 2023-01-06:
Breakdown this setup into smaller one and debug one by one to get an idea what actually is causing the issue

i suggest you to not use "V-mapper"
use 2 straight chains and instead of using "V-mapper", use blank i.e. ""

1st make a single chain of 3, 4, ..., 7 panels and see if everything is working fine
then append the 2nd chain and observe the result 

also i suggest to initially put brightness=100 and pwm-bits=1 just for observation purpose

ignaciogaticarojas, 2023-06-03:
Thanks @ledvinap thats very helpful ! was trying to make an economic change on the adapter that happen to be expensive! appreciate your help

ignaciogaticarojas, 2023-06-03:
@ledvinap I think thats not for the problem since my panels are not E panels. it seems to me that probably has to do that i built a adaptor missing the  third chain chip and resistor. Would not adding this affects the clock signal of the middle chain? @hzeller 

Thanks a lot

davemaster, 2023-06-03:
looks crazy, but, test chains of two, I mean, two by two... from the beginning to the end. If they works fine, is a power supply issue, don't believe in the nominal power in the sticker on power supply. 

Try this, if all panels are right, at the input and output, test power supply, start with one, two, three, etc, then find when the glitch appears, ADD another power supply, not before you test its nominal power.

Best regards

ignaciogaticarojas, 2023-06-05:
@davemaster  Thanks Dave, i tried what you suggested but i saw no improvement. It also seems that the flicker get worst within time. Is maybe because i did not fully complete the active adapter with the third chain chip and resistor, so this week i will build one and try it again. just wondering if anyone had build an active adaptor with just two or one outputs...

Hope everyone had a great weekend

Best

anandrajgupta, 2023-06-05:
can you try 
/--led-pwm-bits=<1..11>
/
just to check it's not the computational power limitation
by default this value is 11, but for your use case the value 8 should also work fine
and for debugging purpose i recommend doing this 1 also and run a scrolling text in a single color