hzeller/rpi-rgb-led-matrix #1409: How can I fix this

Bano310, 2022-03-07:
How can I fix this, 
Problem 1: The words are separated.
Problem 2:The sentences are out of order
Problem 1
[image]
Problem 2
[image]

Bano310, 2022-03-10:
@hzeller Hi master, I am using a RGB Matrix + RTC board and it gives me the mentioned problems how can I solve them.

Dblxy, 2022-03-10:
What is the command that you use to run this?

Bano310, 2022-03-10:
> ¿Cuál es el comando que usas para ejecutar esto?

sudo ./text-example -f  ../fonts/cu12.bdf  --led-chain=8

jonasradke, 2022-03-11:
So led slowdown only accepts range from 0-4.
You should try to use the flags —led-cols and —led-rows to set the amount of pixels in a single matrix. Also it does not look like you have 8 panels chained together but maybe they are just not in the photo.

hzeller, 2022-03-11:
In general @Bano310 it would be helpful if you give more information about your set-up: how many panels do you have, which are the sizes of each panel, how have you wired them up (in one chain, in parallel); what is the `RTC` board you are talking about (link).

As @jonasradke said, set the `--led-cols` and `--led-cols` to the size of a single panel, then use `--led-chain` with the number of panels you have chained together.

The artifact you see looks like, you need to play around with `--led-multiplexing` to see if this matches your panel.

(`--led-slowdown-gpio` suggests 0..4, but it actually allows for up to 5 but it is a hidden feature as this usually should not be needed).

Bano310, 2022-03-14:
https://mail.google.com/mail/u/0?ui=2&ik=979753946b&attid=0.1&permmsgid=msg-f:1727277766572273070&th=17f885fa28f86dae&view=att&disp=safe&realattid=17f885ea672793feb591
this board I'm using, and I have two 32x64 daisy-chained panels.
@hzeller @jonasradke

nvt271201, 2022-06-08:
Hi, I'm having the same problem, how did you fix it

closed 2025-11-27 by marcmerlin