hzeller/rpi-rgb-led-matrix #1411: Running in a docker container

Apollorion, 2022-03-10:
I run a home kubernetes cluster and want to add the raspberry pi as a node to said cluster and run this project on that node.
I am struggling to get the python libraries to install within a container. Any suggestions or help I could get with this? I was able to get it into install on my pi directly but its not going so swell in a container.

Currently this is what I have the container running:

```dockerfile
FROM ubuntu:20.04

RUN apt-get update && apt-get install -y git make build-essential python3 python3-distutils \
    && git clone https://github.com/hzeller/rpi-rgb-led-matrix.git \
    && cd rpi-rgb-led-matrix \
    && make build-python PYTHON=$(which python3) \
    && make install-python PYTHON=$(which python3)
```

It keeps erroring here:
```
#5 33.28 x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I../../include -I/usr/include/python3.8 -c rgbmatrix/core.cpp -o build/temp.linux-x86_64-3.8/rgbmatrix/core.o -O3 -Wall
#5 33.29 rgbmatrix/core.cpp:6:10: fatal error: Python.h: No such file or directory
#5 33.29     6 | #include "Python.h"
#5 33.29       |          ^~~~~~~~~~
#5 33.29 compilation terminated.
#5 33.29 error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
#5 33.30 make[1]: Leaving directory '/rpi-rgb-led-matrix/bindings/python'
#5 33.30 make[1]: *** [Makefile:67: build-python] Error 1
#5 33.30 make: *** [Makefile:28: build-python] Error 2
```

Duality4Y, 2022-03-12:
Probably missing python development files try installing python-dev with apt.

Apollorion, 2022-03-13:
> Probably missing python development files try installing python-dev with apt.

Wow EZPZ yup that was it, thank you so much!

closed 2022-03-13 by Apollorion

stevenbrookslv, 2023-06-28:
I have recently taken up trying to get this to work in a docker container, and while it is working I have noticed very bad flicker relative to running the code without the docker container. Did you face this issue?

Apollorion, 2023-06-29:
@stevenbrookslv Its been a really long time since I looked at this project. But I've been running my raspberrypi with this project in a docker container for a few years now without issue.

If you wanna dig through my source code, maybe something will stand out to you? This is my project here: https://github.com/Apollorion/led-project

Duality4Y, 2023-07-05:
> I have recently taken up trying to get this to work in a docker container, and while it is working I have noticed very bad flicker relative to running the code without the docker container. Did you face this issue?

Are you running the program in docker or only using docker to build the images? could it be a debug build vs optimization kind of thing? compiling a binary in the docker container shouldn't really be causing such issues.

maybe you are loading your cpu a lot somehow?