hzeller/rpi-rgb-led-matrix #1418: My code has a bug

Bano310, 2022-03-30:
My code is blinking, how can I remove this blinking.

this is my code
import json
from samplebase import SampleBase
from rgbmatrix import graphics
import RPi.GPIO as GPIO
import time
import sys

GPIO.setmode(GPIO.BCM)
GPIO.setup(21, GPIO.IN)
GPIO.setup(24, GPIO.IN)
GPIO.setup(18, GPIO.OUT, initial=0)

cont =100
plazas =100
class GraphicsTest(SampleBase):
    def __init__(self, *args, **kwargs):
        super(GraphicsTest, self).__init__(*args, **kwargs)
    def run(self):
        canvas = self.matrix
        font = graphics.Font()
        font.LoadFont("../../../fonts/texgyre-27.bdf")
        
        green = graphics.Color(0, 255, 0)
        graphics.DrawText(canvas, font, 83, 24, green,str(cont))

        blue = graphics.Color(0, 0, 255)
        graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
        time.sleep(1)
while cont <= 100:
    inputValue = GPIO.input(21)
    inputValue1 = GPIO.input(24)
    if (inputValue == True):
        cont = cont - 1
        print("Choche entrada " + str(cont))
        time.sleep(.01)
        class GraphicsTest(SampleBase):
            def __init__(self, *args, **kwargs):
                super(GraphicsTest, self).__init__(*args, **kwargs)
            def run(self):
                canvas = self.matrix
                font = graphics.Font()
                font.LoadFont("../../../fonts/texgyre-27.bdf")
        
                green = graphics.Color(0, 255, 0)
                graphics.DrawText(canvas, font, 83, 24, green,str(cont))

                blue = graphics.Color(0, 0, 255)
                graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
                inputvalue = False
            time.sleep(3)       
    time.sleep(1)
    
    if (inputValue1 == True):
        cont = cont + 1
        print("Salida" + str(cont))
        time.sleep(.01)
        class GraphicsTest(SampleBase):
            def __init__(self, *args, **kwargs):
                super(GraphicsTest, self).__init__(*args, **kwargs)
            def run(self):
                canvas = self.matrix
                font = graphics.Font()
                font.LoadFont("../../../fonts/texgyre-27.bdf")
        
                green = graphics.Color(0, 255, 0)
                graphics.DrawText(canvas, font, 83, 24, green,str(cont))

                blue = graphics.Color(0, 0, 255)
                graphics.DrawText(canvas, font, 1, 23, blue, "plazas")
                inputvalue1 = False
            time.sleep(1)       
    
    time.sleep(1)   
    if (cont == 0):
        GPIO.output(18, 1)
        print ("parking lleno" +  str(cont))
        class GraphicsTest(SampleBase):
            def __init__(self, *args, **kwargs):
                super(GraphicsTest, self).__init__(*args, **kwargs)
                self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel")
            def run(self):
                canvas = self.matrix
                font = graphics.Font()
                font.LoadFont("../../../fonts/texgyre-27.bdf")
                LedChain = 4
                
        
                red = graphics.Color(255, 0, 0)
                graphics.DrawText(canvas, font, 0.5, 23, red,"completo")
            time.sleep(.3)
    time.sleep(2)

        
    if __name__ == "__main__":
        graphics_test = GraphicsTest()
        if (not graphics_test.process()):
            graphics_test.print_help()
time.sleep(5)

https://user-images.githubusercontent.com/101106566/160803942-9fac6711-2991-4e79-9026-ae6df2561a59.mp4

xuniuer, 2022-04-01:
Just to remove 

```time.sleep()```

Not necessary to wait, because there is a background thread to update in library.

Bano310, 2022-04-04:
@xuniuer hello good I just have to remove the time.sleeps?

Bano310, 2022-04-05:
@xuniuer @hzeller 
import argparse
import time
import sys
import os

sys.path.append(os.path.abspath(os.path.dirname(__file__) + '/..'))
from rgbmatrix import RGBMatrix, RGBMatrixOptions

class SampleBase(object):
    def __init__(self, *args, **kwargs):
        self.parser = argparse.ArgumentParser()

        self.parser.add_argument("-r", "--led-rows", action="store", help="Display rows. 16 for 16x32, 32 for 32x32. Default: 32", default=32, type=int)
        self.parser.add_argument("--led-cols", action="store", help="Panel columns. Typically 32 or 64. (Default: 32)", default=32, type=int)
        self.parser.add_argument("-c", "--led-chain", action="store", help="Daisy-chained boards. Default: 1.", default=1, type=int)
        self.parser.add_argument("-P", "--led-parallel", action="store", help="For Plus-models or RPi2: parallel chains. 1..3. Default: 1", default=1, type=int)
        self.parser.add_argument("-p", "--led-pwm-bits", action="store", help="Bits used for PWM. Something between 1..11. Default: 11", default=11, type=int)
        self.parser.add_argument("-b", "--led-brightness", action="store", help="Sets brightness level. Default: 100. Range: 1..100", default=100, type=int)
        self.parser.add_argument("-m", "--led-gpio-mapping", help="Hardware Mapping: regular, adafruit-hat, adafruit-hat-pwm" , choices=['regular', 'adafruit-hat', 'adafruit-hat-pwm'], type=str)
        self.parser.add_argument("--led-scan-mode", action="store", help="Progressive or interlaced scan. 0 Progressive, 1 Interlaced (default)", default=1, choices=range(2), type=int)
        self.parser.add_argument("--led-pwm-lsb-nanoseconds", action="store", help="Base time-unit for the on-time in the lowest significant bit in nanoseconds. Default: 130", default=130, type=int)
        self.parser.add_argument("--led-show-refresh", action="store_true", help="Shows the current refresh rate of the LED panel")
        self.parser.add_argument("--led-slowdown-gpio", action="store", help="Slow down writing to GPIO. Range: 0..4. Default: 1", default=1, type=int)
        self.parser.add_argument("--led-no-hardware-pulse", action="store", help="Don't use hardware pin-pulse generation")
        self.parser.add_argument("--led-rgb-sequence", action="store", help="Switch if your matrix has led colors swapped. Default: RGB", default="RGB", type=str)
        self.parser.add_argument("--led-pixel-mapper", action="store", help="Apply pixel mappers. e.g \"Rotate:90\"", default="", type=str)
        self.parser.add_argument("--led-row-addr-type", action="store", help="0 = default; 1=AB-addressed panels;2=row direct", default=0, type=int, choices=[0,1,2])
        self.parser.add_argument("--led-multiplexing", action="store", help="Multiplexing type: 0=direct; 1=strip; 2=checker; 3=spiral; 4=ZStripe; 5=ZnMirrorZStripe; 6=coreman; 7=Kaler2Scan; 8=ZStripeUneven (Default: 0)", default=0, type=int)

    def usleep(self, value):
        time.sleep(value / 1000000.0)

    def run(self):
        print("Running")

    def process(self):
        self.args = self.parser.parse_args()

        options = RGBMatrixOptions()

        if self.args.led_gpio_mapping != None:
            options.hardware_mapping = self.args.led_gpio_mapping
        options.rows = self.args.led_rows
        options.cols = self.args.led_cols
        options.chain_length = self.args.led_chain=4
        options.parallel = self.args.led_parallel
        options.row_address_type = self.args.led_row_addr_type
        options.multiplexing = self.args.led_multiplexing
        options.pwm_bits = self.args.led_pwm_bits
        options.brightness = self.args.led_brightness
        options.pwm_lsb_nanoseconds = self.args.led_pwm_lsb_nanoseconds
        options.led_rgb_sequence = self.args.led_rgb_sequence
        options.pixel_mapper_config = self.args.led_pixel_mapper
        
        
        if self.args.led_show_refresh:
          options.show_refresh_rate =1

        if self.args.led_slowdown_gpio !=4 :
            options.gpio_slowdown = self.args.led_slowdown_gpio
        if self.args.led_no_hardware_pulse:
          options.disable_hardware_pulsing = True

        self.matrix = RGBMatrix(options = options)

        try:
            # Start loop
            print("Press CTRL-C to stop sample")
            self.run()
        except KeyboardInterrupt:
            print("Exiting\n")
            sys.exit(0)
        

        return True

This is the library I use could you tell me what part of the library I have to change, to solve the flickering.IT IS VERY URGENT !!!!

THANKS FOR THE USERS WHO HELP.

xuniuer, 2022-04-06:
Non-professional suggestions: many factors need to be adjusted or considered against flickering. 
 
1. slowdown gpio value - have a try to adjust the value according to your PI model.
2. enough power provision to board
3. proper wire length - shorter wire length preferred

More details in doc. I remember there are some practical tricks mentioned in the discussion group.

xuniuer, 2022-04-06:
Dig more: https://rpi-rgb-led-matrix.discourse.group/t/flicker-when-display-video-with-rpi3b-and-288x128/278/2
this is a good practice.

Bano310, 2022-04-06:
> Sugerencias no profesionales: muchos factores deben ajustarse o considerarse contra el parpadeo.
> 
> 1. ralentización del valor de gpio: intente ajustar el valor de acuerdo con su modelo de PI.
> 2. suministro de energía suficiente para abordar
> 3. longitud de cable adecuada - se prefiere una longitud de cable más corta
> 
> Más detalles en doc. Recuerdo que hay algunos trucos prácticos mencionados en el grupo de discusión.

Hello very good morning(@hzeller @xuniuer ), I think I have not explained well, sorry, I have echo a counter for a parking that subtracts with a button and the other adds, the problem is that the program is always in loop and so blinks just want to start again when you get a signal from any of the buttons and if you have no input pulses to be kept in standby.

I tried to do it only with (IF) removing the (while) but the code did not execute constantly, it read it once and stopped executing.

Greetings from Spain

Thank you very much for your help. @xuniuer 

https://user-images.githubusercontent.com/101106566/161912601-0995a687-16d7-4379-b1eb-8c3d56fbd9f7.mp4

https://user-images.githubusercontent.com/101106566/161912605-408bb6ef-93c2-4489-8d9e-a1d25c48961a.mp4

xuniuer, 2022-04-06:
Now I understand your problem - when you check a GPIO value then to change content accordingly, it might cause a blink.

My advice is to use gevent which is a micro-thread library instead of builtin thread if Python. The library is more simpler to use and control. The most important, it's light. And, to use wiringpi library to access GPIO.

The following is not complete but has shown how to use gevent and wiringpi libray.

```
# your_gpio.py
import wiringpi

def gpio_init():
    wiringpi.wiringPiSetup()
    # init your PIN
    # ...

def read_gpio(pin):
    return wiringpi.digitalRead(pin)

def write_gpio(pin, val):
    return wiringpi.digitalWrite(pin, val)

# main.py
import gevent

text = ""
def task_check_gpio():
    global text
    # to change your text here
    while True:
        # to change your text here according your gpio
        gevent.sleep(0.1) # sleep 100ms here then to check again

def task_led_matrix_show():
   global text
   # to show your text
   while True:
      # use RGBMatrix to show contents to screen
      gevent.sleep(0.01)

if __name__ == "__main__":
    tasks = []
    task = gevent.spawn(task_check_gpio)
    tasks.append(task)
    task = gevent.spawn(task_led_matrix_show, text)
    tasks.append(task)
    gevent.joinall(tasks)
```

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin