hzeller/rpi-rgb-led-matrix #1421: 64x64 Panel Making Buzzing Noise w/ adafruit bonnet

aidanlincolnn, 2022-04-04:
Hello, big fan of this repository, have made 7-8 led art objects using it with various 32x32 and 32x64 screen chains. I am running into a new issue that I hope someone can help with. 

I just purchased a 64x64 matrix and got it wired up with the Adafruit RGB Bonnet with the E-Line connected to pin 8 (the new Bonnets have a jumper so you no longer need to do the ground plane exacto hack). The panel plays content just fine, however there is a faint buzzing noise coming from the raspberry pi. I am worried there is some overloaded component but don't know how to investigate further. 

Has anyone else noticed this happening? I tried switching out all the wires, a new Pi, a new bonnet, as well as different 64x64 screens (all of them trigger the buzzing), 32x32 or 64x32 panels don't cause any issue on the same pi / bonnet. 

Any help would be greatly appreciated!

This is what I am working on:https://vimeo.com/687232967?embedded=true&source=vimeo_logo&owner=88640089

hzeller, 2022-04-04:
I'd say mostly make sure to have the Raspberry Pi or the Bonnet is not dealing with voltage glitches of the panels.

So I'd have one 5V power supply just for the panels (and connect them with as short as possible and as thick as you have cables from the power supply to the panels), and have a separate 5V power supply just for the Pi (e.g via USB).  It might well be that the panels otherwise create high-frequency voltage fluctuations on their power and if the Pi power is also derived from that, it might have trouble keeping its internal voltages constant.

aidanlincolnn, 2022-04-24:
@hzeller that did the trick, thank you!

closed 2022-04-24 by aidanlincolnn