hzeller/rpi-rgb-led-matrix #1423: Extreme Large Font not working!

anandrajgupta, 2022-04-12:
Hi, Thank you Mr. Henner Zeller...

I am using Raspberry Pi to Make a Large Display of LED Matrix that can has to be mounted at a very large viewing distance.
So, I am making larger fonts using the tool suggested by you: otf2bdf 
The problem I am facing is that Small Fonts are being drawn but Larger Fonts (around 100 pixels high) are not!
There is some sort of Pixel limit per character, because in Larger fonts also few characters are being drawn and few are not, interesting to see that the character which occupies larger number of pixels are getting missed when drawing.

Any help will be highly appreciated!

hzeller, 2022-04-12:
Yes, currently the width is at most 64 pixels (internally, each row is read into a [64 bit bitmap](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/bdf-font.cc#L33)).

I might need to adapt that a bit to be able to deal with wider fonts. Can you share the BDF file you created ?

anandrajgupta, 2022-04-13:
Please give me a hint how can i try to make it support wider fonts also...
Thanks!

hzeller, 2022-04-16:
If you do a git pull now, it should be possible to show wider fonts.

closed 2022-04-16 by hzeller

anandrajgupta, 2022-04-16:
wow thank you sir, i will check it, thank you so much you are so kind!

anandrajgupta, 2022-04-19:
Tested it, working perfectly! Thank you very much!
Just one comment:--
I had to to change the line number 189
from "y_pos = y_pos - g->height - g->y_offset;"
to "y_pos = y_pos;"
because it was taking some negative offset in Y-Axis for the characters!

Thank you again...

commit 39953b7a (2022-04-16): Allow for fonts with glyphs wider than 64 pixels.

Fixes #1423