hzeller/rpi-rgb-led-matrix #1425: Several rows not turning on 32x32

Niek09, 2022-04-13:
So I"m trying to connect a 32x32 HUB75B (don't know what the B indicates) to an electrodragon bonnet. This bonnet already came with the fixed version, as there was a problem with it in February.

The raspberry pi is running DietPi with sound turned off.

So basically what happens is that there are 8 rows on, then 8 rows off, then 8 rows on and 8 rows off. I already tried some multiplexing options, but this didn't work. Does anyone know a solution for this?

[image]
[image]

Please don't mind the cables, there will be 6 panels parallel, but they all have this problem.

ZoyshM, 2022-08-24:
Have you tried to change the --led-row-addr-type flag?

Niek09, 2022-08-25:
The problem was the electrodragon bonnet. I got a new one and the issue was resolved.

closed 2022-08-25 by Niek09