hzeller/rpi-rgb-led-matrix #1426: Font's Lowercase Characters shifting upward automatically!

anandrajgupta, 2022-04-20:
Hi, Thank you Mr. Henner Zeller...

I am using otf2bdf font creator to make various sized fonts, The issue I am facing is that when we write Uppercase characters then everything is fine but when writing Lowercase characters they are shifted upward automatically which makes them look weird as shown in below picture:
[image]

Note: The same issue is not observed when using one of the fonts provided with this library e.g. 8x13B.bdf

Any help will be highly appreciated!

hzeller, 2022-04-20:
Do you have the BDF file for me ? I think it might happen if the font decides to choose a different baseline for each glyph.

anandrajgupta, 2022-04-20:
Yes sir, here it is:
[testFont.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/8519224/testFont.zip)

anandrajgupta, 2022-04-20:
Sir how to configure/modify the font in order to make it use same baseline for each glyph?

Thank you...

hzeller, 2022-04-20:
So the text `Ready` looks ok for me on my screen.

anandrajgupta, 2022-04-20:
The issue has been solved sir!
I had to tick the checkbox saying "don't trim glyph bitmaps" at the time of exporting font from the windows tool I'm using i.e. "Fony", as shown in below screenshot:

[image]

Now, I just have an additional step involved after making any new font using otf2bdf, i.e. feed the otf2bdf generated font into this "Fony" tool then export by unchecking that trim checkbox and the font looks as expected!

In your case you are saying "Ready" looks ok on your screen then maybe there is some other mystery behind it! while I am also using latest updated library. Anyways, I have found a way to overcome the problem I was facing so I'm happy!

Once again, Thank you very very much for your kind support!

Closing this issue now!

closed 2022-04-20 by anandrajgupta