hzeller/rpi-rgb-led-matrix #1429: Help avoiding overlapping text

fofusion, 2022-04-27:
Hi,

I'm making fairly good progress getting this great library to work, but I'm completely lost about how to prevent overlapping static/scrolling text and wondering if anyone has been able to figure this out.

Basically, I want my board to flip through various things such as weather/train times etc and I'm trying to have some scrolling text from right to left which some static text before it in the first 1/3rd of the row.

Eg:
Calling at: <<< SCROLLING SECTION RIGHT TO LEFT >>>

With 'Calling at' being static text.

The issue I have is that the scrolling text overlaps the static piece until it spills over the edge of the display.

I guess the key to this is to find a way to have the text start 'erasing' itself a pixel at a time once it reaches a certain part of the screen so that it doesn't overlap the 'Calling at' section.

Has anyone had any joy with something like this?

My code is [here](https://pastebin.com/PuKMtd8q).

Appreciate any help!

kylejohnsonkj, 2022-05-09:
Draw a rectangle underneath the static text (in my case, the pause button), but above the scrolling text layer. I made it red for this example, but you'd match it to your background color.
[image]

fofusion, 2022-05-11:
Thanks for the reply @kylejohnsonkj !

Forgive my ignorance, but I presume we're talking about setting up pygame and installing the pgmatrix library so that I can utilise pygame with this adapter to draw out areas on screen?

Does this generate the code I require or would I then be running it from pygame?

Sorry, I've not utilised this before and I'm not seeing much in the way of resources online to help me get to that stage.

Thanks!

kylejohnsonkj, 2022-05-11:
Sorry to confuse. You can use the PIL library to draw shapes. Here's a rectangle for example.

```
from PIL import Image, ImageDraw

frame = Image.new("RGB", (64, 64), (0,0,0))
draw = ImageDraw.Draw(frame)
draw.rectangle((0,0,32,32), fill=(255,0,0))
matrix.SetImage(frame)
```

That pygame window is just the output of a matrix emulator I use to make development easier. It's a drop in substitute for this library. Here's the link if you're interested: https://github.com/ty-porter/RGBMatrixEmulator

kylejohnsonkj, 2022-05-11:
Sorry to confuse. You can use the PIL library to draw shapes. This example should be helpful:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/image-draw.py

That pygame window is just the output of a [matrix emulator](https://github.com/ty-porter/RGBMatrixEmulator) I use to make development easier. It's a drop in substitute for this library.

fofusion, 2022-05-31:
Thanks for the link, that's super helpful. I'll take a look

(Sorry for my slow reply!)