hzeller/rpi-rgb-led-matrix #1431: P10 16x32 Outdoor 2014G 1/4 Scan not working

Xaver88, 2022-05-07:
Is there anybody using a P10 Outdoor display 1/4 scan from Novastar and get it work?
I got nothing on the display
tried many combination
if i connect an other display to the same raspberry with active connection board, it works
but not on the P10
[image]
[image]

Xaver88, 2022-05-09:
it seems to have the MBI5043 S-PWM Chip
anyone success with this chip?

xuniuer, 2022-05-20:
P10 is, generally, Hub12 interface, the library just supports Hub75.

closed 2025-11-27 by marcmerlin