hzeller/rpi-rgb-led-matrix #1433: Pins????

Bano310, 2022-05-11:
Hi all, I was wondering which pins are free on the board to be able to use two push button inputs and one output, which pins are ideal for this type of function?

@hzeller @yarikoptic @jwheare @jpoppe

gingters, 2022-05-12:
Great question, I have the same ;)

I did compare the wiring diagram here (https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#connections-per-chain) with the RasPi pin layout, and figured that _all_ data pins are used and only GND/power pins are unused.

However I see that the pins GPIO 5,6,12,13,19 and 20 are used for chain 2 exclusively, the same goes for GPIO 2,3,14,16,21 and 26 which are used for chain 3 exclusively.

That brings me to the actual question:

If I don't use all chains, can I use the pins for the not-used chains independently?
Especially, if I use active buffering board, since I don't know what that thing does with these pins?

Bano310, 2022-05-12:
> Gran pregunta, tengo lo mismo ;)
> 
> Comparé el diagrama de cableado aquí ( https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#connections-per-chain ) con el diseño de pines RasPi, y pensé que _todo_ se usan pines de datos y solo los pines GND/power no se usan.
> 
> Sin embargo veo que los pines GPIO 5,6,12,13,19 y 20 se usan para la cadena 2 exclusivamente, lo mismo ocurre con los pines GPIO 2,3,14,16,21 y 26 que se usan para la cadena 3 exclusivamente.
> 
> Eso me lleva a la pregunta real:
> 
> Si no utilizo todas las cadenas, ¿puedo utilizar los pines de las cadenas no utilizadas de forma independiente? Especialmente, si uso una placa de almacenamiento intermedio activo, ya que no sé qué hace esa cosa con estos pines.

I solved my question on this page, it explains which pins are free and how each of them works.
https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/pinouts

gingters, 2022-05-12:
This is for another HAT that can only drive a single chain. I have the HAT that can drive up to 3 chains.

xuniuer, 2022-05-20:
While two chains ( parallels ) are used, some PINs could be free to use. If the 3rd parallel is used, less PINs are free, at least the default `pi-shutdown-pin` could not be used.

Serzhs, 2023-03-26:
Hello, I am using Raspberry 1.2 and adding `led-no-drop-privs` option helped me to accesses GPIO pins after LED Matrix have been initialised.

Example:
```
sudo python ./show_text.py  --led-gpio-mapping=regular-pi1  --led-cols=64 --led-rows=64 --led-no-drop-privs
```

Serzhs, 2023-08-24:
I encountered an issue with GPIO pins. When I ran the LED matrix, I couldn't retrieve data from my sensors, which were utilizing the remaining available GPIO pins. The root cause of this problem was the outdated Raspberry Pi model I was using. It simply lacked the necessary resources to simultaneously read and write data to both the 64x64 LED panel and the DHT22 Sensor. To address this, I transitioned from using the **Raspberry Pi 1 B** to the more capable **Raspberry Pi 4 4B** model, and now everything functions smoothly.