hzeller/rpi-rgb-led-matrix #1435: 64x64 RGB LED-Matrix not working properly (Split Image)

jan424242, 2022-05-26:
Hey, I have a Project for School and my 64x64 RGG LED-Matrix is not working properly (image)
We are using a Adafruit HAT with RTC

The command is: sudo ./demo -D 0 --led-rows=64 --led-cols=64 --led-gpio-mapping=adafruit-hat
Does anyone have similar problems? I need help asap, thanks :D

[image]
[image]

hzeller, 2022-05-26:
Did you do the necessary hardware mod on the HAT that is described in the README? https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet

jan424242, 2022-05-30:
Yes, I have a newer version of the "HAT" than described in the "Readme" file. With this HAT, you have to put a dot between E and 8 next to the HUB75 connector. (you can see it on the Image)

kaerka, 2022-08-21:
I have the same panels, and same hat, and am having the same issue.  I've nearly exhausted all options - it seems like maybe either the code or the hat just can't handle ABCDE/64x64 panels, or perhaps an issue with the 1/32 scan rate?

Mine shows a nearly identical set of black bars across it, along with flicker - but only when on the 64x64 smaller LED panels.

I have:
- tested the panels with other setups, including the matrixportal m4 and pimoroni i75 - the panels work and don't have this issue.  (Those, however, don't have enough memory to run what I want, however.)
- tested dropping from 8 panels to 4, then to 2, then to 1 - same issue even on a single panel.  
- swapped from older adafruit hat to newer version with the E pin mod, but without the audio pin mod (yet)
- reflowed all of my solder, checked continuity and solder points.
- swapped to a dedicated power supply (a 300w x 5v - 60a psu)
- retested the same hat/pi4/code with 2 x 16x32 older panels (no issues there)
- tested with different IDC cables - no change

My setup is (at the moment, it changes as I test different things):
- pi4, 8gb (no OC, disabled 1-wire, enabled I2C and SPI), wifi connected
- Raspbian 10 (buster)
- Commit ID: f88355e46faafb9de7f1e59dd258ab36c7e7b097

Anyway - on my setup, I'm not sure where to go with further troubleshooting - but am willing to help test and post photos/videos of results.

kaerka, 2022-08-21:
Here are a few photos of mine, and also my command line for those.  Ignore the panel on the right and below, those have different code running on them.

$ sudo ./demo -D 0 -m 100 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 --led-brightness=50 --led-chain=4 --led-rows=64 --led-cols=64 --led-multiplexing=0 --led-pixel-mapper="U-mapper:1" --led-row-addr-type=0

[image]

$ sudo ./demo -D 9 -m 150 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 --led-brightness=50 --led-chain=4 --led-rows=64 --led-cols=64 --led-multiplexing=0 --led-pixel-mapper="U-mapper:1" --led-row-addr-type=0
[image]

kaerka, 2022-08-21:
I reflowed my E pin behind the connector, and then also did the jumper for the pins #4 to #18; and then recompiled for that.  The reflow of my E pin seems to have been the fix for me - now it's working excellently!  I just love your code - thanks and awesome!
[image]
[image]

ignaciogaticarojas, 2022-09-06:
Hi ! @kaerka could you please share images of your hardware mod on the hat?

Thanks!

kaerka, 2022-09-06:
Sure - I had to reflow the solder on the DE pin behind the hub75 connector several times before it worked correctly, and then I added the jumper between pins 4 and 18, and recompiled for the quality option.

I'm still doing other things to improve the rate that I can push it, but it's pretty stable now.

This is the pins you have to short for a 64x64 panel to get those lines to go away, assuming that is the issue.

[image]

These are the pins for higher quality display output (it mainly reduces a little bit of flicker and artifacts).  My jumper wire is atrocious, but it is working.
[image]

And here is a picture of the overall 8 x 64x64 panel setup.  I also upgraded my power supply, though at a different step.  The 5v x 15a PSU I had was not enough, once I got it all working.
Ignore the couple extra 32x32 panels, they're on separate controllers.
Front:
[image]

Back:
[image]

Better macro focus of the overall hat on a pi4:
[image]

ignaciogaticarojas, 2022-09-06:
Hi @kaerka that is awesome, can you please explain me what do you mean by "reflow" the DE pin, does this mean removing it?
Thanks a lot and congrats on such a nice display

kaerka, 2022-09-06:
Thanks!  Most of the thanks really goes to the developers of the rgb-matrix code and adafruit - most of this is beyond me, and I'm using it as a fast-paced python learning experience.

Reflowing the solder is to basically re-heat it, such that it resettles and provides a better connection.  I'd have thought that my first attempt was good - it looked fine... but it didn't work.  It took a total of 4 times re-flowing to get it 100% correct, I think it was overlapping onto the 3rd contact a bit, and then later I could see a little line between the two contacts which were supposed to be jumped.  The little line between them indicated that they weren't really, even if the solder looked connected.

I did figure out that when trying to get it to re-flow - be patient, give it a a few min for the iron to heat up correctly and don't try to rush it.  And then, also be careful that you're only putting heat on the solder and pads, and not on the board itself.  

Last - in my case, this wasn't the only solder joint that I figured out that I'd done poorly.  I reflowed many of the gpio pins as well.  They don't look great, but all are showing good continuity on a multi-tester, so seem fine now.

combs, 2023-05-07:
nice! Probably makes sense to close this issue out then?

closed 2023-05-07 by hzeller

davemaster, 2024-05-24:
> Hi ! @kaerka could you please share images of your hardware mod on the hat?
> 
> Thanks!

I had a Electrodragon adapter, and was very easy to mod it, because is a build-in connector paths to do it. Pin E to pin 8, or pin E to pin 4. 

[image]

In my case, I welded a 3-pin header for quick set or unset option

[image]

Here, the final result, in picture:

[image]

Here a [video of my adventure today](https://www.youtube.com/watch?v=sBRi85QoGuY)... mission, acomplished.