hzeller/rpi-rgb-led-matrix #1436: Flickering when using c# code, gpio slowdown higher than 5?

mrmrmystery, 2022-05-26:
Hi,
I wrote a program that displays an image inside an 2d array. I set the pixel to the matrix with 2 simple for loops. However, There are random pixels flickering in Different colors. I found out, that higher gpio slowdown reduces the flickering. Is there an option to go beyond  5? What do I have to change?

rotrico, 2022-06-29:
Hi mrmrmystery,
I had same problem with different panels, rpi4 is too fast event at slow 5.
I suggest you a workaround: underclock rpi4, edit /boot/config.txt file and find those lines:

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

Uncomment arm_freq line and put anything between 700 (minimum, lower values could prevent raspberry to boot) and 1500 (default frequency).

whatleymd, 2023-07-21:
Did you ever find a solution to this? arm_freq is at 800 and I'm passing in 5 as the gpio slowdown as part of the RGBLedMatrixOptions struct

closed 2025-11-27 by marcmerlin