hzeller/rpi-rgb-led-matrix #1439: Not divisible by 2 - Mapping 9 panels

ggggggr, 2022-06-01:
I have a 1 daisy chain setup of (x9) 64x64 adafruit rgb led matrix panels, how can I make it happen?

The terminal tells me to make them ¨--led-chain=9 has to be divisible by 2¨,  it displays a bad mapping of my file. I tried connecting 3 more, to make it 12 panels (divisible by 2), with the same parameters + ¨--led-chain=12¨ and it doesn't map properly. The panels are fully working, they display the pulsing colors demo on each panel. 

sudo ./led-image-viewer /mnt/usb/192.gif --led-cols=64 --led-rows=64 --led-gpio-mapping=adafruit-hat-pwm --led-chain=9 --led-pixel-mapper=U-mapper

Thanx!

xuniuer, 2022-06-13:
I bet panels with U-mapper/one chain/one parallel needs even number because the chain will be folded, like U, to display. To solve it, use three parallels with 3 panels one chain with default mapper, if 9 panels are used, or to use 8 or 10 panels with one parallel with U-mapper to see proper display.

davemaster, 2022-06-13:
Had You tried to define 10 panels (1 ghost panel)?

Good luck

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin