hzeller/rpi-rgb-led-matrix #1442: Support for FM TC5020A chipset?

ZoyshM, 2022-06-14:
I had been trying to figure out why my panel wouldn't display things properly, and I realized that it uses an unsupported chipset. The image it produces seems somewhat similar to what it should look like however, as the attached video is what demo 0 looks like on it. I am using a pi 3 with regular mapping, it is a 64 col, 32 row panel, and I have gpio slowdown of 2. Increasing or decreasing the gpio value only makes the panel display total gibberish so that is a dead end. Has anybody made a panel with this chipset work?

https://user-images.githubusercontent.com/98439408/173474564-559f7e8e-8d42-4378-a908-96916d3668df.mp4

ZoyshM, 2022-08-24:
For anyone who ever runs into this problem, change the --led-row-addr-type flag. Changing it to 4 fixed the issue entirely for me.

closed 2022-08-24 by ZoyshM

anandrajgupta, 2022-08-25:
Great,
That must be completely dependent on type of panel being used ✌🏻