hzeller/rpi-rgb-led-matrix #1445: I cant use a P5 Panel

saintgemi, 2022-07-11:
Hey i have a problem with the library and panels i bought. Previously I used P5 64x32 1/8 panels and everything worked perfectly. I bought the same panels again and it turns out that when I connect them they don't look the same as before, when I looked at them more closely I realized that they don't have the address line D. I already tried all the ways, changing the multiplexing, etc, and it still doesn't work, if someone has an idea I can try

Steveraws, 2022-07-12:
try --led-row-addr-type=3 most new led matrix panels use shift registers for colom or row selection. just go trough all --led-row-addr-type and --led-multiplexing parameters

saintgemi, 2022-07-12:
i already tried with --led-row-addr-type=3 with --led-multiplexing=1 and 0 and i tried with a multiplexing mapping made by other user and nothing. I have 5 chains and 3 parallel, all the panels are p5 1/8 without D address line. Before i used the same configuration but p5 1/8 with D line and that configuration doesn't work with this new's panels

closed 2025-11-27 by marcmerlin