hzeller/rpi-rgb-led-matrix #1452: New Adafruit 32x64 4mm Panel not working on Raspberry PI

resmithn, 2022-08-12:
I just purchased an Adafruit 32x64 4mm matrix panel. I have an existing setup: RaspberryPI 3B+ running Scrolling Sign [BigjBehr]. That works fine, been running for couple of years. When I tried the new panel on the existing setup all I get is scrambled lines. [Picture attached]. I tried the panel on an Arduino 2560 running simple scrolling script, works okay. Same with Adafruit Matrix Portal running OpenWeather, works okay. Contacted Adafruit support and they had no solutions and suggested this site. Also tried another Raspberry PI 3B+, downloaded RGB Matrix Master and Scrolling Sign programs, works okay with an existing 32x64 panel but not the new one.
Adafruit support mentioned about Chips FM6126A and FM6127. The chips on the new panel have RUC7258D.
Appreciate thoughts on how to resolve this issue.
Thank You.
[image]
[image]

ZoyshM, 2022-08-24:
Try changing the --led-row-addr-type. My panel was flickering and displaying complete gibberish but when I changed the addr type to 4 it worked perfectly.

resmithn, 2022-08-24:
I am running the BigJBehr Scrolling Sign, in the samplebase.py only options
are 0, 1 & 2.
Thanks for your reply!
self.parser.add_argument("--led-row-addr-type", action="store", help="0 =
default; 1=AB-addressed panels;2=row direct", default=0, type=int,
choices=[0,1,2])

ZoyshM, 2022-08-24:
I didn't realize the Python bindings supported one fewer choice for addr type. Try using addr type 4 for one of the C++ example demos and see if that works.

hzeller, 2022-08-25:
The current code does contain the led-row-addr-type 4

https://github.com/hzeller/rpi-rgb-led-matrix/blob/5e3fb5f9fdc3d1f41e21dd4f46fa6bff0218d13a/bindings/python/samples/samplebase.py#L28

If you don't have this flag with that version, your repo is about 2 years old @resmithn (it was added in [Feb 2020](https://github.com/hzeller/rpi-rgb-led-matrix/commit/4c8274e92008dbe0bc3efe52c64a924007cab3ae)).
If you git pull, you should have that option available.

ZoyshM, 2022-08-25:
This makes a lot more sense, it is still not guaranteed to fix the issue but it is at least worth a shot.

resmithn, 2022-10-11:
> Are you using an adafruit RGB matrix hat/bonnet? I had a similar issue
> because I wasn't selecting the adafruit-hat gpio mapping (e.g. with the
> demo: sudo ./demo --led-cols=64 --led-gpio-mapping=adafruit-hat -D 0

I have RPI's set up with both Hat and Bonnet. I can get the ./demo to run
okay with a two year old Adafruit 32x64 panel, but plugging in the one I
just purchased, I get scrambled lines as described. Same with Scrolling
Sign [BigjBehr <https://www.instructables.com/member/BigjBehr/>], the old
panel works okay, the new one has the scramble lines. Thanks for your reply.

closed 2025-11-27 by marcmerlin