hzeller/rpi-rgb-led-matrix #1453: Help after exhausting options for an 128x64 ABC panel

taintedkernel, 2022-08-12:
Hello, I'm having difficulty getting an 128x64 ABC panel to function without artifacts/glitches.  I've also been testing a chain of 4 64x32 displays and after identifying a number of issues I finally have these working without problems, but the native 128x64 panel still has problems.

I'm using an Adafruit RGB Matrix Bonnet on a new RPi 4 with the anti-flicker hardware mod.  Recorded 2 videos, one of the artifacts on the native panel and one with the 4 panel combo working flawlessly - https://photos.app.goo.gl/5c3Et18sRQHtYzeq8

Things I've tried:
- Larger 5V PSU @ 10A
- Soldered low-ESR caps to the 5V power lines at the connector to PCB
- Tried older v4 kernels since I was aware of other issues with GPIO speed on newer kernels (https://github.com/raspberrypi/linux/issues/3381)
- Moved from Raspbian to DietPi (this helped a great deal)
- Tinkered with all available options in the software (led-pwm-lsb-nanoseconds=50 had the best results)
- Other tips also done (disabled sound, no GUI, isolcpus=3, etc)

The only other thing that stuck out to me was the mention of a low logic level voltage, but the Adafruit adapter says it has level shifters onboard.  The HUB75 data cable is only about 5" long.

As mentioned, the hardware works flawlessly (now) with a set of 4 displays giving me a combined 128x64, so I think the problem is with the larger panel or limitations with the ABC addressing.

Thanks for this project and your time!  Appreciate any suggestions :smile:

istvanovics, 2022-10-18:
I have the same problem. Rpi4 with DietPi + Electrodragon HAT( I tried the older and the newer versions too).

taintedkernel, 2022-10-22:
Update - I just discovered today that I think the issue here was with the LED panel this whole time.  I purchased another panel of the same type (128x64) and couldn't tell at the time, but it also happened to have ABC addressing.

Finally got a chance to try it out, and it seems to work flawlessly despite not doing any of the other things I mentioned above (except using DietPi).  Maybe this will help you out @istvanovics but I'll close out the issue.

taintedkernel, 2022-10-27:
Actually scratch that, I remembered the specs incorrectly - the new panel that works for me is actually the more-traditional ABCDE addressing, so I think this issue still stands.  Reopening it to see if we can tackle things.

juansource, 2022-12-20:
Hi there, been a while since the last update on this issue, but I am having the same glitching problem with a single 128x64 panel that I got from [Aliexpress](https://www.aliexpress.us/item/3256801498872813.html). I am also using a Adafruit matrix bonnet and followed their [guide](https://learn.adafruit.com/adafruit-rgb-matrix-bonnet-for-raspberry-pi/). I don't know what type of board I have, whether it's ABC or something else but I am willing to figure it out. I've done some of the troubleshooting steps from this post and istvanovics' [post](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1472) with modifying the config files for the core allocation, deactivating the sound module, increasing power supply amperage from 2A to 10A, soldering/shorting the E port for the 8 multiplexing but no luck yet. Curious if anyone has come up with a solution yet.

taintedkernel, 2022-12-20:
@juansource I think I have the same panel on my end, if so it's an ABC (you can tell from the lack of D & E address lines labeled on the HUB75 connector).  No update on my end here, but I have another similar panel that I've been meaning to try out to see the results.

juansource, 2022-12-22:
@taintedkernel I see, from the looks of it, it seems like I [have](https://imgur.com/a/c9dpCRn) an ABCDE-type panel. After doing some research over the past few days, I have discovered that there is a [solution](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1340) that involves putting a jumper cable between the 4 and 18 ports of the Adafruit bonnet, this has been called the PWM hardware fix. I am not an expert in desoldering, so I put a jumper cable between the two and got a slight improvement, but not fixed entirely. But what I am thinking now is that it might come down to the chipset of the board. Because I have been seeing several posts asking if their specific chipset works, and I have only found one [post](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1452) that mentions one of the two exact chips that I [have](https://imgur.com/a/8G9bFnM) (RUC7258, the other is ICN1037BP) but they were also inconclusive with their results even with hzeller's input. This is what my panel is looking like now with the sample demo. [Image](https://imgur.com/a/sMmUuuL). So I am thinking of just finding a board that has been verified to work. Thanks for the help!

davemaster, 2022-12-23:
> Hi there, been a while since the last update on this issue, but I am having the same glitching problem with a single 128x64 panel that I got from [...]

Greetings,

Using ONE PANEL, have You tried use wires instead your module? Try it first (an advice)

juansource, 2022-12-24:
@davemaster Hello! After connecting the one panel directly to the Raspberry Pi (no bonnet) and double checking the wires, this actually fixed the problem! There is still some very mild glitching, but it is way better than it was before. Thank you so much! Saludos!

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828