hzeller/rpi-rgb-led-matrix #1458: Image and Video viewers failing to make.

Kwesst, 2022-08-23:
Hi all, attached is a photo of me installing the dependencies as written in the readme.

However I am still unable to make led-image-viewer or video-viewer. Text-scroller works fine. Also image-example also fails in a similar fashion to the attached. When trying to execute video-viewer I get “Issue opening file: : No such file or directory” which could also be my fault. I’m trying to play videos on a mounted usb, so I’m not sure how detailed I need to be leading the video-viewer to the video file. 

I did some due diligence and went through all issues back to 2016 and noticed a few others with the same problem, but no fixes. One user said that apt update and upgrade fixed a kernel issue, but this did not solve the problem for me. 

I’m using a pi4 running dietpi with a single panel for testing. 

At best two lines flashed on screen during the make process, but after repeated attempts, same process, even those are no longer showing up. 

Thanks in advance for any help or diagnoses. 

[image]

hzeller, 2022-08-23:
it looks like it actually made the led-image-viewer.

What happens if you invoke it with

```
./led-image-viewer
```

Kwesst, 2022-08-23:
Thanks so much for the quick reply. 

“expected image filename” along with readme

hzeller, 2022-08-23:
Perfect, so it works :)
Just give any image (and the usual --led-rows, --led-chain, --led-parallel etc that you'd give for your set-up) and it should work.

Note, that the image viewer reads file as user 'nobody' (unless you give the `--led-no-drop-privs` flag), so your image needs to be readable to everyone. `chmod a+r myimagefile.png` if it is not.

Kwesst, 2022-08-23:
Ah, wow the —led-no-drop-privs flag oh my goodness. 

Wow wow wow big thanks! 

Let me know if I can cashapp you something small for your time!

closed 2022-08-24 by Kwesst

anandrajgupta, 2022-08-24:
Few things are not done for money!
We should come to post only after completely reading the readme and testing all the possibilities, it will save a lot of people's time.
Thank you so much Mr. @hzeller.